EndPointCorp/lg_ros_nodes #106: Spherical pano

wzin, 2015-10-12:
Development notes:

- we need equirectangular pano images to be able to project them onto a sphere. This is already done (link needed @mvollrath )
- we need to convert single image to cube map (N,E,S,W) - crop and distort chops of image
- single large image cannot be loaded initially on the sphere - there needs to be asynchronous way of loading the details
- check sv_pano.js

mvollrath, 2015-10-13:
This issue is missing any reference to actual requirements, while working with Jacob he mentioned that we need to support zoom.  This is incompatible with the GSVPano approach because it loads a single texture limited to about 4096 width which is 1/4 actual Street View panorama detail and very blurry when zooming.

I dug up the old krPano Street View node server and we ported the application to hook into /streetview/ topics via ROSLIB.  Presently pan and tilt are working, navigation is doing things but there seems to be some heading discrepancy.  Performance and presentation are pretty good.

Known issues, things to do:

* Diagnose/fix navigation heading shift (@mvollrath can diagnose, fix)
* Store pov/pano state on server, replay on connection (@mvollrath)
* Support zoom (@minshallj server side, @mvollrath client side)
* Find out what else needs to be done, prioritize (@wzin et al.)
* Performance testing vs. LGX

To run it:

* Start roscore, spacenav_node, rosbridge websocket, and lg_sv server.py (no params needed)
* Install node 10.x
* Clone the [lg_kr_streetview repo](https://github.com/EndPointCorp/lg_kr_streetview).
* Enter the repo and `npm install`
* Start the server with `node index.js`
* Point a browser at <http://localhost:8086/display/> (will start black)
* Publish a pano message: `rostopic pub -1 /streetview/panoid std_msgs/String ZphZajKeR_lJjBJftY0SYw`
* You should be able to drive around with the SpaceNav.

wzin, 2015-10-14:
Deployment notes for krpano based SV:
- drop node js server on headnode
- use lg_sv launcher to point browsers to: http://lg-head:8060/display?yaw_offset=N

mvollrath, 2015-10-15:
Jacob and I worked on the THREE.js version, it has been pushed to <https://github.com/EndPointCorp/lg_sv_nonfree> with instructions for running the PoC.

The krPano stack has been updated today with cleanup, more thorough setup doccage, and cbkproxy compatibility with the THREE.js client.

wzin, 2015-10-15:
We need to demo it **before** 21st (ideally 20th) of October.

mvollrath, 2015-10-15:
Moving dev discussion to https://github.com/EndPointCorp/lg_sv_nonfree/issues/1

closed 2015-10-28 by wzin