EndPointCorp/lg_ros_nodes #111: change streetview panos on spacenav button click

minshallj, 2015-10-29:
Shouldn't be hard to implement. There's probably a rostopic somewhere to listen in on this, and it shouldn't be too hard once we find that to add this into the sv_server.

Requested by @bengoldstein

mvollrath, 2016-02-24:
This was merged, but there's an issue where, while in Earth, pressing a button on the Nav causes the street view pano to move.  This triggers a flyto the new pano location.  We need to ignore button presses while street view is inactive.

Thanks @mababio

minshallj, 2016-02-24:
Oh, shucks. I can probably fix that up real quick.

minshallj, 2016-02-25:
Fixed in 6730557389dfa435a276163a0d6406c22c548071. Just needed to wait until the state is true (visible).

minshallj, 2016-02-25:
tested with @mababio and the fix worked well. Closing this even though the fix won't be in a new deb package until a new release is created.

closed 2016-02-25 by minshallj