EndPointCorp/lg_ros_nodes #113: Support for LG offline mode

wzin, 2015-11-03:
Since a majority of LG ROS nodes depend on internet connectivity - we need an **ultra** reliable connectivity detector that will turn LG to offline mode by a configurable action (most likely show "sorry - no internetz" placehoder on the screen)

Requirements:
- implement different check methods - ICMP, resolve DNS name, HEAD request with curl
- template the hostname - try many hostnames
- if no internet is detected - emit director message
- provide configurable director message on per-installation basis

Davejenk1ns, 2015-11-03:
This will only help us if it provides precise information:
"The Liquid Galaxy/Portal is functioning normally, but there is no connection to the Internet.  Please contact the network administrator for this office."

My guess is that they'll still just reflexively ping end point.

zdenekmaxa, 2016-07-12:
further notes based on a discussion earlier today:

- need portal-ros connectivity ros node 2.0 better in aspects
  - number of probed metrics to determine connectivity (perhaps controled by a node definition ros configuration list)
  - configurability of actions driven by messages
  - reliability -> QA

- will have tests - by means of mock library to emulate online / offline network status
- logic in the python module (against which to run both online and offline ros tests) rather than in a script like the connectivity node has

@wojo - anything to add for the first iteration on this issue?

wzin, 2016-07-13:
>number of probed metrics to determine connectivity (perhaps controled by a node definition ros configuration list)

not only number but also type:

>ICMP, resolve DNS name, HEAD request with curl

right now it's only HEAD request I think

zdenekmaxa, 2016-07-13:
yeah, that's what I meant. a configurable list of metrics.
yes, just HEAD now.

zdenekmaxa, 2016-07-14:
next development notes:

- configurable everything
    - list of metrics, addresses tried within metrics
    - actions, messages emitted upon detecting offline status (not clear yet)

- there will be a service informing whether the system is offline, online

- portal-ros connectivity did just HTTP HEAD request - sometimes some sites block it, also here, doing stuff repeatedly from several tens of systems, some local admins may notice robotic traffic -> need to investigate methods of reliably establishing network status on frequent basis and need to have more metrics (initially the equal weight) and doing the round of checks several times (2-3) before pronouncing offline status ; equally ICMP may sometimes be blocked, etc ; emulate browser agent behaviour in HTTP request?

- don't use any 3rd party libs for DNS, ICMP communication. use CLI tools and suprocess process start with exit status checking, if necessary command output evaluation (e.g. ping -W <timeout>)

wzin, 2016-08-09:
Moving to proper pipeline.

zdenekmaxa, 2016-08-10:
any other feedback @wojo?

wojo, 2016-08-10:
Wrong person ;)

zdenekmaxa, 2016-08-10:
oh, sorry @wojo :)

zdenekmaxa, 2016-08-10:
any other feedback @wzin ?

zdenekmaxa, 2016-08-11:
I believe all functionality requirements are fulfilled by the above commit. Unittests (both online, offline) are done. Everything is ROS node def configurable.
Going to deployment-test on the system, then lg_ros_nodes release and a new ISO roscoe release to full chain deployment test.

zdenekmaxa, 2016-09-07:
- more configurables implemented, now everything is configurable
- the sensitivity can be tuned via number of last rounds of checks to consider for offline status
- lg_chef databag ros def values updated too

zdenekmaxa, 2016-09-07:
portal-ros connectivity ros node review:

max_failed_attempts - default 5. 
the lg_offliner defines number of rounds of checks to consider before claiming offline status. this is a slightly different approach, but the same functionality still can be achieved by defining number of rounds of checks = 1. then each of these commands needs to fail to announce offline status. Having number of rounds higher would result into lower sensitivity.

lg_offliner is a lot more versatile and configurable when it comes to particular network commands to check the connection with as well as configurability of online, offline messages.

I am well confident that lg_offliner achieves identical functionality compared to portal-ros connectivity plus a lot more just by different configuration variants adjusted to the network quality.

cc @wzin

closed 2016-09-14 by wzin

zdenekmaxa, 2016-09-22:
do not use nslookup as one of the metric commands. on some system the response might be returned from DNS cache even when the system is factually offline (and lg_offliner would erroneously think - since 1 command from the commands list succeeded - that system is online).

just a matter of removing the check command from the ros node configuration.

zdenekmaxa, 2016-09-22:
notes, references for later integration:

https://github.com/EndPointCorp/lg_chef/issues/471 offline page (cookbook)
https://github.com/EndPointCorp/lg_chef/issues/927
https://github.com/EndPointCorp/ros_cms/issues/564

lg_ros_nodes branches (work underway):
        topic/set_initial_state
        topic/uscs_initial_state