EndPointCorp/lg_ros_nodes #118: video assets that are the same should not be restarted

minshallj, 2015-11-18:
This might be tricky, and I'm not totally sure where this fix would need to be implemented.

Labeling regression since this worked in previous systems (non ros based)

wzin, 2015-11-24:
Hmm - I think I have explicitly set it to be restarting when new message comes from the director. Why shouldnt we restart playback?

minshallj, 2015-11-24:
If you have one video that you want to persist while assets change around
it. Maybe images on the side giving further detail to what's being talked
about in the video.

wzin, 2015-12-18:
Makes a lot of sense to me. 

This would involve configuration here: https://github.com/EndPointCorp/ros_cms/issues/123 because I imagine we won't always want to do it like that. 

I also think it's already implemented.
If we publish the same URL (even with different geometry) but URL stays the same - it's [not going to be restarted](https://github.com/EndPointCorp/lg_ros_nodes/blob/development/lg_common/src/lg_common/adhoc_browser_pool.py#L77-L104). I think what we need is to check whether js part reloads the video.

@minshallj it surely doesnt work like that for mplayer (which is obsolete).

mvollrath, 2019-07-09:
@minshallj it seems like the launcher handles this now?

minshallj, 2019-07-09:
I believe so. If the coordinates / size hasn't changed, and the URL stays the same assets are left alone

closed 2019-07-09 by minshallj