EndPointCorp/lg_ros_nodes #124: Add support for loading Chrome extensions to ManagedBrowser

wzin, 2015-12-17:
Requirements:
- make managed browser accept a list of extensions
- make it part of metadata attribute in [interactivespaces_msgs.GenericMessage](https://github.com/EndPointCorp/lg_ros_nodes/blob/development/interactivespaces_msgs/README.md) e.g. 
```json
{
"metadata": {
  "chrome":{
    "extensions":  ["path1", "path2", "path3"]
 }
}
```
-

Dev notes:
- [ManagedBrowser class](https://github.com/EndPointCorp/lg_ros_nodes/blob/development/lg_common/src/lg_common/managed_browser.py#L27)

minshallj, 2016-01-19:
After looking into this for a bit, it seems like the only way to load extensions via the command line is to use unpacked extensions. Packed extensions can be installed via the command line, but it seems to take user interaction plus I'm not sure how quick the installation would be if we could find a policy to avoid user interaction.

If we do use unpacked extensions, they'll need to make their way to the display nodes somehow since I haven't been able to load them over http (yet)

minshallj, 2016-01-20:
[image]

wzin, 2016-02-29:
If it's not merged (yet) then please attach PR here.

minshallj, 2016-02-29:
pr in #163

closed 2016-08-31 by wzin