EndPointCorp/lg_ros_nodes #126: Influxdb statistic ROS node

wzin, 2015-12-17:
Make support for a ROS node that will listen on /director/scene and submit scene names to influxdb. It would also retrieve scene name from another ros node about current scene for better stats resolution.

Requirements:
- write ROS node A that extracts `slug` of [interactivespace_msgs.GenericMessage](https://github.com/EndPointCorp/lg_ros_nodes/blob/development/interactivespaces_msgs/README.md) published on `/director/scene` and submits it to director
- write ROS node B that remembers last scene and provides service with last known scene
- make ROS node A ask ros node B about last known scene and submit stats so we get configurable resolution of stats

zdenekmaxa, 2016-01-20:
notes pertinent to this development:

- stats in the context of portal/roscoe merge:
https://github.com/EndPointCorp/lg_chef/issues/484

- "stats are hard", FD discussion:
https://www.flowdock.com/app/endpoint/r-d/threads/-4SpGGhw_8d8Mr3YR8tSuvY7Jz4

- extent of this ticket: write a ROS node that will listen on some (configurable) ros topics and extract attribs from messages and submit it to influxdb via telegraf submission agent

- this node will be common for both roscoe and portal. portal is depended on appctl but also on packages built from lg_ros_nodes, chain:
https://github.com/EndPointCorp/lg-live-image/blob/portal/config/package-lists/portapt.list.chroot
here's what gets installed on portal - first package is a metapackage from portal-ros repo, second and third are coming from lg-ros-nodes repo

- this new (lg_stats?) ros node will be included liquidgalaxy metapackage

- topics to sniff on, message types, time resolution - everything configurable via ros params

zdenekmaxa, 2016-03-30:
@wzin from the above spec, there are the following particular points I'd like to discuss further:

- /spacenav/twist - not clearly defined above

- average proximity sensor distance

- touchscreen touch events processing

- both 1 message memory and time resolution parameters are implemented, however, e.g. in case of /directory/scene watcher, there won't ever be submitted any stats message based on this topic since the same message is never sent twice (with len(buffer) = 1) so nothing will ever be submitted into stats

- I'd appreciate a separate ticket for the systems ros node B (e.g. FPS submission) and more details about the desired metrics

The work is done otherwise and I am working on testing and deployment.

wzin, 2016-03-31:
> /spacenav/twist - not clearly defined above

Discard messages that consist of zeros and count non-zero messages during resolution period (e.g. 30s).

> touchscreen touch events processing

touchscreens' touch events are not available in ROS - you need to configure lg_replay to attach to your touchscreen and listen on a topic that it provides.

Submit message rate to telegraf.

> Proximity sensor message

It's not nested (but **prepare for nested messages** )

> both 1 message memory and time resolution parameters are implemented, however, e.g. in case of /directory/scene watcher, there won't ever be submitted any stats message based on this topic since the same message is never sent twice (with len(buffer) = 1) so nothing will ever be submitted into stats

- submit stats item containing slug on every /director/scene topic message
- re-submit the last stats item every 30s (or stats_resolution s)

use two different measurments for above ^

>I'd appreciate a separate ticket for the systems ros node B (e.g. FPS submission) and more details about the desired metrics

let's not do it right nao.

wzin, 2016-03-31:
Next steps:
- make it deployable, merge to master and create a release
- assume that we are not using complex messages (only one layer deep slots)
- prepare for submitting following types of data:
 - average proximity sensor distance
 - spacenav twist (discard zeros)
 - lg_replay/touchscreen messages (rate of messages)
 - director messages (as events + resbumission of last message)

zdenekmaxa, 2016-03-31:
deployment:
 - merge to master
 - update https://github.com/EndPointCorp/lg_ros_nodes/blob/development/liquidgalaxy/package.xml metapackage so that new ros node lg_stats is picked up
 - release making: https://github.com/EndPointCorp/lg_ros_nodes/blob/development/README.md

- template lg_ros node in lg_chef
- make a note that there is just 1 lg_stats ros node per system

- more /director/scene notes:
   - submit every incoming non-empty (need to have a check for that) message caught (slug as event)
   - submit last received messages every some_update_parameter_name seconds (even if it's at some point stop-presentations message, just keep submitting it all over again ...) (via a background thread?)
   - if there is nothing ever received, don't do anything

zdenekmaxa, 2016-04-15:
- lg_ros_nodes release 1.1.46 made, include lg_stats deployable build
http://portapt.galaxy.endpoint.com:5831/lg-ros-indigo-1.1.46/pool/main/r/ros-indigo-lg-stats/

- verified deployment on portaldevcz system, /director/scene watching and stats submission works as expected

zdenekmaxa, 2016-04-15:
outstanding items:

prepare for submitting following types of data:
- average proximity sensor distance
- spacenav twist (discard zeros)
- lg_replay/touchscreen messages (rate of messages)

these shall IMO become separate tickets and close this ticket then.

as a prerequisite, finish this:

https://github.com/EndPointCorp/lg_ros_nodes/issues/187

lg_stats depends very much on it. tests are missing there and also the functionality which may be necessary for watching average values or rates of some of the three topics above.

zdenekmaxa, 2016-04-22:
here are some comments to the recent work @wzin :
https://github.com/EndPointCorp/lg_ros_nodes/commit/0ee15a45e215365e37b2e05459a98ec63386ed2a

1. other than the comments below, it looks good

2. as already mentioned in the commit message, there are no tests in the lg_stats for the functionality added 

3. equally there are no tests for lg_common/src/lg_common/helpers.py on which data driven lg_stats ros node depends so much (btw, this is already tracked here: https://github.com/EndPointCorp/lg_ros_nodes/issues/187)

4. the issue https://github.com/EndPointCorp/lg_ros_nodes/issues/200 was implemented, ticket closed

5. lg_ros_nodes development jenkins projects has been failing since the yesterday's commit
[image]
Although it's suboptimal leaving the tests implementation for later in general, leaving the projects failing on CI is wrong, sorry. This needs fixing 9 out of 10 lg_stats test cases, though it's all just variables names ...

6. what was the motivation for having two ros_nodes defined in chef? lg_stats and lg_stats_portal ? I see just the watched topics difference. If that is solely the case, then /director/scene is useless in the lg_stats_portal definition and shall be removed.

7. from the TODO comment: "remove thread spawn as it's not safe to use them inside ROS script
instead use rospy.sleep() and rospy.Duration() or rospy.Rate()" I think we'll need to elaborate more on the idea. Alas, I was unable to find why exactly is this code not thread-safe or rospy recommendation on thread-safety in this case.

8. lg_replay/touchscreen messages (rate of messages) is currently the only outstanding issue (provided that above points are satisfied, tests fixed / implemented). And if this topic is no longer considered, then this monster ticket is done.

zdenekmaxa, 2016-04-22:
from @wzin 

value field of Event needs to be float32. so we need two message types instead of one.
let's say StatsEvent.msg and StatsMetric.msg. first used for events like mode changes

wzin, 2016-04-22:
1 ok
2 and 3  we don't have time for it now 
5 we will fix tests if time allows 
6 testing 
7 discard it - no longer valid 
8 i would use count strategy  here - should be easy to configure 

thanks for putting my fd private comments here zdenek!

zdenekmaxa, 2016-04-22:
"fix tests if time allows". I gotta memorize that. Just beautiful.

more actions items from @wzin FD discussion:

- confirm that LG replay works with this ROS node today -eg try to record count of touchscreen touch events [comment: there is no configuration for this topic in the ROS node def, so I am assuming it has not been tested and thus it's pretty safe to assume (e.g. given the state of the tests) that this feature is dysfunctional.]

- based on the float32 vs string comment above, divide Event to two messages - one with string,  the other with float32

- connect Event (containing string value) to "default" strategy. the other message type should be connected to "count" and "average"

- confirm that using rate and count in the same context is correct

zdenekmaxa, 2016-05-10:
- the lg_stats tests have been failing on jenkins since yesterday

- lg_stats implementation-wise, having multiply methods with identical if-else clauses strongly suggests to redesign and involve inheritance. in other words, average, count, rate should become own processor child classes

@wzin

wzin, 2016-05-20:
Closing the ticket as lg_stats initial release is functional since 1.2.8

closed 2016-05-20 by wzin

zdenekmaxa, 2016-05-20:
I see making code-review comments makes little sense here ^^. It's closed and buried for ever now. Noted.