EndPointCorp/lg_ros_nodes #127: Implement Onboard keyboard ROS node

wzin, 2015-12-17:
We need an [onboard keyboard](https://apps.ubuntu.com/cat/applications/precise/onboard/) that will be ran on every viewport using common channel e.g.: `/onboard/state`. These unique onboards will accept individual `show` requests coming from viewports that touch input is being sent to. 

Requirements:
- make support for running one onboard instance per viewport - let onboard have appropriate size depending on viewport size
- make all instances listen on common channel `/onboard/state`
- use message type with action and viewport name e.g.:
```
str viewport
bool active
```
- let all onboards perform "hide" action each time a message is sent to onboard instance other than them
- by design - onboard show/hide messages should be sent by client applications (client = application run inside of a container like google-chrome)

szymonlipinski, 2015-12-18:
Scripts for the current onboard ros node implementation: https://github.com/EndPointCorp/portal-ros/tree/development/catkin/src/onboard/scripts

Basically to show/hide onboard we need to send a dbus message: https://github.com/EndPointCorp/portal-ros/blob/development/catkin/src/onboard/scripts/listener.py#L20-L26

There is also a dbus function call which returns the onboard state

wzin, 2016-08-10:
Ask @mvollrath to know how it's implemented in portalterminator.

zdenekmaxa, 2016-09-08:
@mvollrath - could you please drop some references here pertinent to onboard implementation in portalterminator? thanks.

mvollrath, 2016-09-08:
For terminator there were a few changes:

* A separate launcher node was factored out.
* Instead of keeping onboard running all the time, it is spun up on demand.
* The keyboard is coerced into the bottom of its viewport.

There is more to do to integrate it with mirroring.
* Need to run a node for each viewport.
* The node needs to be aware of the touch routing state so it can control visibility correctly.

In other words, each viewport must have a configured onboard node, but only the active touch viewport will show on a visibility state rising edge.

wzin, 2016-09-20:
from converstaion with @mvollrath - onboard should follow touch router

wzin, 2016-09-21:
Some notes from the meeting:

- tactile or any other rosbridge-connected webapp may publish `std_msgs/Bool` on `/onboard/visibility`

- develop onboard router ros node that will listen on `/onboard/visibility`
 - make it listen on director messages as well to decide on which viewport onboard should be spun on
 ```json
 {"activity_config": { 
 "route_touch": true,
"viewport": "your_viewport"
  }
 }
 ```
 - it needs to have a default viewport that it shows onboard on (in most cases `route_touch` won't be set
 - make onboard router publish list of show/hide events for "your_viewport" (or more viewports) on `/lg_onboard/activate`
 - run only one instance of lg_onboard router

- make an executor/actuator ros node that will subscribe to `lg_common/StringArray`:`/lg_onboard/activate` and will run on every viewport
 - make it show onboard on a viewport that it takes care of: start + applicationstate.VISIBLE with geometry

Scenario:
- user taps text field on the touchscreen
- the fields emits the message on `/onboard/activate` (it's in the kiosk extension)
- router remembers last director message - it has a default of `kiosk` but target viewport can be overriden with `route_touch`: true and `viewport` : `foobar`  in `activity_config` 
- router publishes `["kiosk"]` on `/lg_onboard/activate`
- kiosk lg_onboard executor spins onbaord and shows it on kiosk

Question:
- how do we show 3 onboards on a single displaynode on 3 separate viewports
(this may not be possible at all and we dont need it on portal but the answer may be important for later development)

zdenekmaxa, 2016-09-30:
thanks @mvollrath for your yesterday's fixes - I confirm onboard shows and hides OK on portaldevcz. There is a problem with hiding when displayed on the wall, which I am looking into.
I have also continued with some refactoring and shortened things.

closed 2016-10-03 by wzin