EndPointCorp/lg_ros_nodes #128: Implement storing all custom attributes in "metadata" attribute

wzin, 2015-12-17:
Child of https://github.com/EndPointCorp/lg_chef/issues/463

[interactivespaces_msgs.GenericMessage] has following structure: https://github.com/EndPointCorp/lg_ros_nodes/blob/development/interactivespaces_msgs/README.md

We've been adding fields (or attributes) to window(s) objects but it's not a good idea. We want to keep unstructured and optional data in "metadata".

Proposed message structure is as below. It starts mirroring mode by capturing viewport named "center_portrait_three" to viewport named "touchscreen". It also starts a `3240x1920` browser on "center_portrait_three" that receives touchscreen touches and shows onboard keyboard. Browser is pointing to https://youtube.com

```json
{
  "description": "",
  "duration": 0,
  "name": "mirror mode",
  "resource_uri": "",
  "slug": "mirror-mode",
  "windows": [
    {
      "activity": "browser",
      "assets": [
        "http://youtube.com"
      ],
      "metadata": {
        "keyboard": {
          "enabled": true
        },
        "touch_input": {
          "enabled": true
        }
      }
      "height": 1920,
      "presentation_viewport": "center_portrait_three",
      "width": 3240,
      "x_coord": 0,
      "y_coord": 0
    },
    {
      "activity": "mirror_streamer",
      "assets": [
        "viewport://center_portrait_three"
      ],
      "metadata": {
       }
      "height": 1920,
      "presentation_viewport": "center_portrait_three",
      "width": 3240,
      "x_coord": 0,
      "y_coord": 0
    },
   {
      "activity": "mirror_streamer",
      "assets": [
        "mirror_streamer://42-a/center_portrait_three"
      ],
      "metadata": {
       }
      "height": 1920,
      "presentation_viewport": "center_portrait_three",
      "width": 3240,
      "x_coord": 0,
      "y_coord": 0
    },
  ]
}
```

mvollrath, 2015-12-17:
What is the use case for this metadata field?

mvollrath, 2015-12-17:
Keyboard and touch input should not be attributes of windows, but of viewports.  Also, they will never not be attributed to the same viewport.

wzin, 2015-12-18:
@mvollrath metadata is for telling client running in the browser "yes you can be sending onboard `show`/`hide` messages using your viewport" by passing it in GET arguments that will propagate down to e.g. onboard and/or touch_input extension(s). Onboard extension running google-chrome on viewport center_three_portrait would emit "show" to onboard running on "center_three_portrait" and also would request for sending touch events to "center_three_portrait" by sending message liek:
```
viewport: center_three_portrait
command: activate
```
to a node listening on dispnode that's reponsible for managing this viewport. This node would stop ignoring messages published on `/touch/events` by touchscreen node and would start relaying them to Xorg by writing to uinput device.

wzin, 2016-01-07:
Updated message from the meeting:

```json
{
  "description": "",
  "duration": 0,
  "name": "mirror mode",
  "resource_uri": "",
  "slug": "mirror-mode",
  "activities": [
    {
      "activity": "browser",
      "assets": [
        "http://youtube.com"
      ],
      "activity_config": {
      },
      "height": 1920,
      "presentation_viewport": "center_portrait_three",
      "width": 3240,
      "x_coord": 0,
      "y_coord": 0
    },
    {
      "activity": "mirror",
      "assets": [
      ],
      "activity_config": {
        "source": "center_portrait_three"
      },
      "height": 1920,
      "presentation_viewport": "touchscreen",
      "width": 1080,
      "x_coord": 0,
      "y_coord": 0
    }
  ]
}
```

raw notes:

- evdev teleport receiver logic:
 - react on every emssage that has 'mirror'
 - "mirror": "presentation_viewport" is the source
 - "mirror":"height" + "mirror":"width" become xinput source that need to be translated "browser":"height", "browser": "width" taken from the window that contains viewport defined in "activity_config":"source"
 
- application that's captured needs to cover whole viewport

1080x1080 -> 3240x1080 - evdev teleport receiver - call xinput to make matrix transformations

for input:
 - xinput -> 3240x1080 -> 100x200+50+50
 - https://github.com/EndPointCorp/lg-live-image/blob/portal/config/includes.chroot/etc/X11/Xsession.d/45x11-custom_lg-xinput#L34-L43

eggyknap, 2016-01-27:
The activity_config seems like a good place to put the "extended media player attributes" mentioned in https://github.com/EndPointCorp/ros_cms/issues/123

wzin, 2016-02-29:
Issue moved to [EndPointCorp/ros_cms #302](https://github.com/EndPointCorp/ros_cms/issues/302) via [**ZenHub.io**](https://www.zenhub.io/)

closed 2016-02-29 by wzin