EndPointCorp/lg_ros_nodes #131: Implement chrome page monitor

wzin, 2015-12-18:
Requirements:
- make a parameter liek `monitor_page_url`
- provide list of allowed urls regexps with wildcard using numbered list of urls liek `allowed_url_1=http://localhost:8088/*` and `allowed_url_2=http://localhost:8089/*`
- check chrome list of tabs periodically (parametrize it) to check if the URL matches the regexp

axisofentropy, 2015-12-18:
On 12/18/2015 12:05 PM, Wojciech Ziniewicz wrote:
>   * check chrome list of tabs periodically (parametrize it) to check if
>     the URL matches the regexp

Would this check tab crashes? Does that change the URL?

ezekiel, 2015-12-18:
When the tab crashes, it simply disappears from the "chrome list of tabs" being checked periodically. So this _will_ catch tab crashes if at least one of the allowed URLs is required to be matched AND only one tab is opened per browser instance.

In python, for example, _requiring_ the "set" iterable returned by:
```
<current_page_urls>.intersection(<allowed_urls>)
```
to be non-empty, would catch tab crashes where only one tab is open.

This _will_NOT_ catch tab crashes otherwise.

closed 2016-10-03 by wzin