EndPointCorp/lg_ros_nodes #135: roslaunch dies on headnode

minshallj, 2015-12-29:
Roslaunch has been dieing on our headnode. It's leaving an error that looks like this:

```
roscore/master is not yet running, will wait for it to start
master has started, initiating launch
... logging to /var/log/ros/706c0244-ae65-11e5-9d52-10bf484d0042/roslaunch-lg-head-777.liquid.glx-26033.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://lg-head:57694/

SUMMARY
========

PARAMETERS
 * /rosdistro: indigo
 * /rosversion: 1.11.9
 * /viewport/center: 1080x1920+1920+0
 * /viewport/left_one: 1080x1920+0+0
 * /viewport/right_one: 1080x1920+0+0
 * /viewport/touchscreen: 1920x1080+0+0

NODES

ROS_MASTER_URI=http://lg-head:11311

core service [/rosout] found
No processes to monitor
shutting down processing monitor...
... shutting down processing monitor complete
```

Mostly the last few lines look suspicious. Some places on the internet point to errors in nodes, but there are no nodes in the roslaunch file. This seems to be happening on different versions of rosnodes.

wzin, 2016-01-04:
blank roslaunch xml file will always make roslaunch exit because there's nothing to monitor. 

Moved to `in development` as it seems to be a quickie.

wzin, 2016-01-18:
@minshallj anything you would like to implement here?

closed 2016-01-28 by wzin