EndPointCorp/lg_ros_nodes #136: custom chrome flags

minshallj, 2015-12-30:
Don't kill me @mvollrath :)

But we need to add custom chrome flags to adhoc managed browser and managed browser.

Use case: cross site origin problems. Adding --disable-web-security and --test-type resolves this issue.

For now a simple boolean to add those two flags as a pair will be added, but eventually we might want to add any custom chrome flag.

axisofentropy, 2015-12-30:
This is no longer required for Barclay's "Insights" app (but it'd still be nice to have in the future.)

wzin, 2016-01-04:
Please don't implement it @minshallj. It's a serious security threat.

Instead of removing CORS security, we may proxy everything thru lg-head lighttpd/nginx proxy without problems.

wzin, 2016-01-04:
Moved to "in development" column as you are probably working on it #babysitting

closed 2016-01-21 by minshallj