EndPointCorp/lg_ros_nodes #137: kmlsync logs too much

minshallj, 2015-12-30:
Maybe there should be a debug mode that could allow the debugging, but other than that there's no need to print out every time we get a request on the server. The logs are getting large.

wzin, 2016-01-04:
+1 - moved to ready for backlog. This should be moved to ready for development soon.

wzin, 2016-01-28:
Fixed - all loginfos were kinda irrelevant and low-priority. I moved them to logdebug and one got moved to logerr.

closed 2016-01-28 by wzin

minshallj, 2016-01-28:
Did that reduce the size of the logs? Or just re-classify them?

wzin, 2016-01-29:
@minshallj it reduces amount of rosout logs whcih are producing uncompressable 100-200MB per day in our case because default verbosity for rosout_agg topic which is causing the problem is LOG. Once you switch to DEBUG wil get tons of logdebug messages that you (probably) want.