EndPointCorp/lg_ros_nodes #140: Prevent duplicate KML loading in Earth

axisofentropy, 2016-01-11:
We saw this on the two Barclay's staging systems, `testtn` and `devtn`.  The Queens KML GroundOverlay persisted only on the `center` display even after unloading the Scene.
[image]
This is probably because the ROS node was telling it to load this asset twice, but only unloading once.

					<NetworkLink id="http___lg_head_8088_roscoe_assets_asset_storage_barclays_ada_queens_tour_kml">
						<name>http___lg_head_8088_roscoe_assets_asset_storage_barclays_ada_queens_tour_kml</name>
						<Link>
							<href>http://lg-head:8088/roscoe_assets/asset_storage/barclays_ada-queens-tour.kml</href>
						</Link>
					</NetworkLink>
					<NetworkLink id="http___lg_head_8088_roscoe_assets_asset_storage_barclays_ada_queens_tour_kml">
						<name>http___lg_head_8088_roscoe_assets_asset_storage_barclays_ada_queens_tour_kml</name>
						<Link>
							<href>http://lg-head:8088/roscoe_assets/asset_storage/barclays_ada-queens-tour.kml</href>
						</Link>
					</NetworkLink>

This is probably a bug within Earth, but that's not going to be fixed, so our ROS node should prevent duplicate asset names from being loaded.

mvollrath, 2016-02-24:
I wouldn't assume that this is an Earth bug.  kmlsync is pretty hairy.  We should write an automated test that hits kmlsync with about 100 add and then 100 remove requests simultaneously and go from there.

mvollrath, 2016-03-07:
Also seen: KML not loaded where it should be.

wzin, 2016-07-22:
This got fixed.

closed 2016-07-22 by wzin