EndPointCorp/lg_ros_nodes #150: Specify language in GE labels

mikemerrill, 2016-01-25:
There is a need to be able to specify the language showing on the labels in GE. Currently the default seems to be English, with no mechanism for changing to anything else.

wzin, 2016-01-26:
This may be already working fine with `DHCP_LANG` persona var in `lg-live-image` which is [here]. This persona var setting may have to be ported to `roscoe` branch.

Another option would be a possibility to dynamically specify environment variables (like `LG_LANG`) for processes to be able to set language of labels.

For now - let's port the language setting from [lg-live-image](https://github.com/EndPointCorp/lg-live-image/blob/azero-raring-earth32/config/includes.chroot/home/lg/etc/shell.conf#L19) back to `roscoe` branch.

FuriousJulius, 2016-01-26:
I've added support for DHCP_LANG for the roscoe branch in the roslaunch init script [here](https://github.com/EndPointCorp/lg-live-image/commit/1ce4bc73e54546d6296e9753f6aac37304b95c23).  My work to include this in lg_persona is complete but hasn't been merged.

axisofentropy, 2016-01-29:
Is this merged now?

FuriousJulius, 2016-01-29:
The roscoe image changes and lg_persona stuff is all in place.  https://github.com/EndPointCorp/lg_ros_nodes/pull/151 needs to be merged into lg_ros_nodes and we should be good to go!

FuriousJulius, 2016-03-08:
#151 was merged successfully.  Closing this issue

closed 2016-03-08 by FuriousJulius