EndPointCorp/lg_ros_nodes #177: VNC connections over 42-a:5900 not issuing mouseClicks

b-berry, 2016-03-15:
To whom it may concern,

Ops and Content teams have noticed within the last week that frequently VNC sessions to 42-a:5900 on ros systems lack mouseClick functionality.  The result is a view-only remote desktop experience rather than the desired interactive experience.  

@axisofentropy mentioned possible ngTouch regression or version change of some sort.  

https://www.flowdock.com/app/endpoint/azero-client/threads/tFJxNPqVA6pTbaodH6Cu4bYwRz5

axisofentropy, 2016-03-23:
This commit breaks VNC clicks on the `default` touchscreen https://github.com/EndPointCorp/lg_touchscreens/commit/7460826e52894eaaa3a02852928f8b23f55b14e6

Next I'll step through each release of `angular-material` and hopefully find out which one changed.

axisofentropy, 2016-03-23:
Issue moved to [EndPointCorp/lg_touchscreens #77](https://github.com/EndPointCorp/lg_touchscreens/issues/77) via [**ZenHub.io**](https://www.zenhub.io/)

closed 2016-03-23 by axisofentropy