EndPointCorp/lg_ros_nodes #178: Clear Earth cache during relaunch (Add option)

b-berry, 2016-03-15:
Hello,

On `lg_cms` galaxies running lg-relaunch would clear out:

`42-*:/home/lg*/.googleearth/{{Cache,Temp,crashfiles}/*,*.km{l,z}}`

I've noticed that I often have to manually `rm !$` (the above) on roscoe systems.  Can we add an option to `lg-relaunch` to include this cleanup filepurge or I guess just do this during `lg-relaunch --full`?

wzin, 2016-03-17:
Note: earch files reside in `/tmp/.<earth_instance_name>` by defalut (that are being cleaned up upon every GE instance spawn) but there are unmanaged artifacts in `/home/lg/.googleearth/*` 

The problem is that /home/lg cache is shared and the /tmp/ cache is per-instance. 

Maybe (?) try to make GE put it's files into non-tmpfs predictable dir and rely on #132 mechanism for detecting whether the instance dir size is too big.

minshallj, 2016-06-01:
@wojo, this does happen on each lg-relaunch and lg-soft-relaunch now, since lg-relaunch first calls the soft relaunch. Is this acting up anywhere?

wzin, 2016-06-01:
nope - this ticket was created Mar 15 :)

closed 2016-06-01 by minshallj