EndPointCorp/lg_ros_nodes #184: Need to configure mplayer to span across multiple connected scenes

tj-endpoint, 2016-03-23:
There is some setting in ROS on EPHQLGXROS that is looping videos.  Let's say we have a scene that's a minute long, and a video that's only 15 seconds.  That video currently will play for 15 seconds and then immediately start over and loop until the scene kicks over to the next in the queue.  This is different behavior from our old system, where a video would play it's designated length and then stop.
The problem we're having is that in the prior version we could ALSO take a 5 minute video, for instance, and span it across multiple smaller scenes, and the video would continue playing as each scene kicked into the next one, as long as all consecutive scenes had that video asset in them.  Now, in the ROS version (or, maybe the ROS CMS), if you have a longer video in two different but connected scenes, as soon as the second scene starts it just kills the video and starts it over again.  

Right now we're working on a presentation that needs video to span multiple scenes.  I know that there are many situations like that out in the wild too, specifically CBRE, which will need to be handled when we migrate their systems.

wzin, 2016-04-06:
I'm closing this ticket and I'll ping you on flowdock

wzin, 2016-04-07:
Requirements:
- loop_video: true | false
- on_finish: kill_window | do_nothing
- don't restart video if the same asset was loaded into mplayer (basically two consecutive director messages contain the same mplayer message/window)
- implement custom attribute in "activity_config": {} - `activity_config` should always stay there - it shoudl be a static attribute and if there's no custom config then it shoul be empty
- for streetview panoid/heading/tilt we need to move the custom data to "activity_config" but it shoudl also remain in "window" attributes for some time until we decide that all clients should already support "activity_config" attribute
- every asset type should have a dropdown with custom attributes list. Let's begin with:
 - for streetview: heading, pano , til
 - for mplayer: loop, action on video finish

Example director message:

```json
{
      "activity_config": {
           "mplayer": {
              "loop": true
           }
       },
      "activity": "earth",
      "assets": [
        "http://lg-head/lg/assets/kml/population.kml"
      ],
      "height": 1920,
      "presentation_viewport": "left",
      "width": 1080,
      "x_coord": 0,
      "y_coord": 0
}
```

closed 2016-04-07 by wzin