EndPointCorp/lg_ros_nodes #189: Sluggish and jerky Pano and SV behavior

tj-endpoint, 2016-04-06:
**Overview:**
Street view and the pano viewer experiences have been slowly degrading over these past few months.  My guess is that updates to various parts of the system have been applied over the months, but dedicated SV and pano viewer testing hasn't been a part of each update.  At present, SV is very jerky and sluggish.  The behavior is worse if you 'peg' the spacenav in a full twist to either right or left as you're attempting to rotate.

https://drive.google.com/open?id=0B_uLPLWkN3AoVmJnWGhHVVk1R2M

Very long load times for panoviewer scenes (able to see from video, sometimes a second, sometimes 3 or more), as well as dramatically jerky and lagged twisting, especially when fully 'pegged' CW or CCW. 

https://drive.google.com/open?id=0B_uLPLWkN3AoQnlnQThtRmFreWM

mvollrath, 2016-04-26:
Thanks all, I've created #203 for the camera smoothness.

Long load times will need more investigation.

BenWitten, 2016-04-28:
We gave a demo today for IMG. They're very interested in the system, and will be bringing the decision maker in the next week or two. This issue will have a big impact on how they feel about the system.

mvollrath, 2016-05-04:
Closing this for now, let me know if there's any more performance problems in sv.

closed 2016-05-04 by mvollrath