EndPointCorp/lg_ros_nodes #190: Inconsistent load time of assets, worse for large, and the system can be easily overwhelmed

tj-endpoint, 2016-04-06:
**Overview:**
Often, loading a large asset file will lag one or more screens many seconds after others populate the asset:

[image]

It isn't terribly hard to overload the system and have assets lagged multiple stages beyond each other on different screens: 

[image]

Crashing often accompanies this kind of behavior:

[image]

wzin, 2016-04-11:
Great data @tj-endpoint - we'll discuss it in dev team and will get back to you. I'll need you to replicate this issue and provide time when it happened to be able to look into system resources.

wzin, 2016-04-18:
@tj-endpoint we'll need lg-version output each time you subit a bug.

wzin, 2016-04-18:
Also - a link to a presentation that's slow is also needed.

wzin, 2016-05-02:
@tj-endpoint is this still persistent?

wzin, 2016-08-23:
Possible solutions:
- long polling
- make research and re-test it

axisofentropy, 2016-08-23:
This ticket is only about Google Earth.

We can't do much about large assets.  But small assets we may be able to load quicker using HTTP "long polling" much like Sean's original PHP implementation.

wzin, 2016-08-26:
Some findings after loading asset of http://sos.noaa.gov/kml/kml/all_sats.kml to all earth instances:
- earth does not load the asset on each instance uniformely
[image]
- it reports back to kmlsync server that it has loaded the asset though
```
[tornado.access][INFO] 2016-08-26 16:37:55,204: 200 GET /network_link_update.kml?window_slug=center&bboxWest=67.66962749624697&bboxSouth=-69.22888287624659&bboxEast=211.5838546912529&bboxNorth=29.83150021401583&lookatLon=139.62674109375&lookatLat=-2
5.4481875859375&lookatRange=5000000.77&lookatTilt=0&lookatHeading=0&cameraLon=139.62674109375&cameraLat=-25.4481875859375&cameraAlt=5000000.77&asset_slug=http___sos_noaa_gov_kml_kml_all_sats_kml (127.0.0.1) 2.06ms
[tornado.access][INFO] 2016-08-26 16:37:55,302: 200 GET /network_link_update.kml?window_slug=right_one&bboxWest=154.2193022624842&bboxSouth=-62.90143442582424&bboxEast=212.0069119234984&bboxNorth=26.91699727136901&lookatLon=-161.8079940902979&lookat
Lat=-13.93789289348594&lookatRange=9422387.68&lookatTilt=90&lookatHeading=68.492&cameraLon=139.62674109375&cameraLat=-25.4481875859375&cameraAlt=5000000.77&asset_slug=http___sos_noaa_gov_kml_kml_all_sats_kml (127.0.0.1) 2.43ms
[tornado.access][INFO] 2016-08-26 16:37:55,464: 200 GET /network_link_update.kml?window_slug=left_one&bboxWest=67.11129771314742&bboxSouth=-62.90554156474339&bboxEast=125.0415627620108&bboxNorth=26.97705110640851&lookatLon=81.06147627779795&lookatLa
t=-13.93789289348594&lookatRange=9422387.68&lookatTilt=90&lookatHeading=-68.492&cameraLon=139.62674109375&cameraLat=-25.4481875859375&cameraAlt=5000000.77&asset_slug=http___sos_noaa_gov_kml_kml_all_sats_kml (127.0.0.1) 2.50ms
[tornado.access][INFO] 2016-08-26 16:37:56,215: 200 GET /network_link_update.kml?window_slug=center&bboxWest=67.66962749624697&bboxSouth=-69.22888287624659&bboxEast=211.5838546912529&bboxNorth=29.83150021401583&lookatLon=139.62674109375&lookatLat=-2
5.4481875859375&lookatRange=5000000.77&lookatTilt=0&lookatHeading=0&cameraLon=139.62674109375&cameraLat=-25.4481875859375&cameraAlt=5000000.77&asset_slug=http___sos_noaa_gov_kml_kml_all_sats_kml (127.0.0.1) 2.82ms
[tornado.access][INFO] 2016-08-26 16:37:56,316: 200 GET /network_link_update.kml?window_slug=right_one&bboxWest=154.2193022624842&bboxSouth=-62.90143442582424&bboxEast=212.0069119234984&bboxNorth=26.91699727136901&lookatLon=-161.8079940902979&lookat
Lat=-13.93789289348594&lookatRange=9422387.68&lookatTilt=90&lookatHeading=68.492&cameraLon=139.62674109375&cameraLat=-25.4481875859375&cameraAlt=5000000.77&asset_slug=http___sos_noaa_gov_kml_kml_all_sats_kml (127.0.0.1) 2.03ms
[tornado.access][INFO] 2016-08-26 16:37:56,469: 200 GET /network_link_update.kml?window_slug=left_one&bboxWest=67.11129771314742&bboxSouth=-62.90554156474339&bboxEast=125.0415627620108&bboxNorth=26.97705110640851&lookatLon=81.06147627779795&lookatLa
t=-13.93789289348594&lookatRange=9422387.68&lookatTilt=90&lookatHeading=-68.492&cameraLon=139.62674109375&cameraLat=-25.4481875859375&cameraAlt=5000000.77&asset_slug=http___sos_noaa_gov_kml_kml_all_sats_kml (127.0.0.1) 2.40ms
```

wzin, 2016-08-26:
The KMZ mentioned in the original ticket is using a KML file with KMZ coming from a site that does not exist anymore: http://satellitedebris.net/satellite/spaceobjects.kmz but the fact of pulling assets from outside of lg-head makes me think that we should find a scene that fails to load assets consistently from a headnode.

wzin, 2016-08-29:
We weren't able to reproduce this problem - closing the ticket.

closed 2016-08-29 by wzin