EndPointCorp/lg_ros_nodes #193: Implement mplayer looping behavior

wzin, 2016-04-07:
**Overview:**
Context https://github.com/EndPointCorp/lg_ros_nodes/issues/184

**Requirements:**
List of requirements that need to be satisfied that are needed to consider this feature implemented.
- make bridge between director messages and AdhocMedia messages interpret "activity_config": "mplayer" : "loop_video" attribute and act accordingly by setting mplayer command line arguments

**Development notes**
Add "loop_video" attribute to AdhocMedia message with default of `false`

**Testing scenario**
Emit Adhoc media with "loop_video" set to true and false and observe the behavior.

zdenekmaxa, 2016-05-19:
after checking lg_media and MPlayerPool implementation, there is nothing which makes a video play  again after it has finished as @tj-endpoint describes on #184.
actually, after emitting a single message:
lg_media/AdhocMedias '[{id: "1", url: "FILEURL", media_type: "video", geometry: {x: 10, y: 10, width: 600, height: 800}}]' --once

the mplayer plays the video and finishes after the video is over, there is no looping.

need to investigate further what causes the looping behaviour or what sends the multiple identical messages which makes it to appear the video is looping

some hint @wzin would be helpful here.

also ManagedMplayer adds cmd.extend(["&"]) to the mplayer command which is not only unnecessary but causes mplayer to complain:

```
Playing &.
File not found: '&'
Failed to open &.
```

it tries to open it as a second URL ... removing it.

Upon the identical lg_media/AdhocMedias message, the video starts from the beginning again, the first one doesn't finish.

eggyknap, 2016-05-24:
The original spec for ROS CMS asked for a boolean value to determine whether the video should loop, and a second value to determine whether the video should close or remain open to the last frame, when it finishes. This has been consolidated in ROS CMS as one attribute with three possible values.

```
    {
      "activity": "video", 
      "activity_config": {
        "loopVideo": "close"
      }, 
      "assets": [
        "http://localhost:8000/assets/asdfsadfa"
      ], 
      "height": 540, 
      "presentation_viewport": "left_one", 
      "width": 960, 
      "x_coord": 71, 
      "y_coord": 621
    }, 
```
The message above asks the system to play the video at `http://localhost:8000/assets/asdfsadfa`, and the "loopVideo" value tells it to close the player when it finishes. "loopVideo" is a bad name for the attribute; we can probably change it more easily now than later when it is more widely used, so I propose changing "loopVideo" to "onFinish". Anyway, it can have one of three values: "close", meaning close the player, "nothing" meaning leave the player open, or  "loop" meaning, obviously, restart the video.

Here's another example:
```
    {
      "activity": "video", 
      "activity_config": {
        "loopVideo": "loop", 
        "panoid": "asdf"
      }, 
      "assets": [
        "http://localhost:8000/assets/Some location"
      ], 
      "height": 960, 
      "presentation_viewport": "left_two_portraits", 
      "width": 1080, 
      "x_coord": 611, 
      "y_coord": 563
    }
```
Note that this has a "panoid" attribute as well. This should be ignored. It's probably not likely that extra stuff shows up in the activity_config section, but it's not (yet) impossible.

zdenekmaxa, 2016-05-26:
my installed ros_cms is: https://github.com/EndpointCorp/ros_cms/commit/f839f22
/srv/ros_cms
 - tag: 1.4.0-sketchfab_test-2-7-gf839f22

I am simply unable to define a video asset and reproduce the looping behaviour. Need to discuss with @eggyknap

zdenekmaxa, 2016-06-07:
@tj-endpoint 

if you check the scenes attributes CMS UI, it defines following options:
  - Do Nothing (the window stays, blank)
  - Restart the video (which means play in the loop until perhaps the scene duration elapses)
  - Close the window

it it just these 3 actions which the mplayer needs to accordingly behave or you have something more in mind?

zdenekmaxa, 2016-06-10:
the reason for mplayer restarting until the scene duration is exhausted is in the following chain:
ManagedMplayer->ManagedApplication->
    appctl-ProcController->ProcRunner while True-start

until the ProcController receives stop/shutdown, the ProcRunner has "whilte True" loop and restarts the process. In other works, respawing a process indefinitely is a default behaviour in appctl/ProcRunner.

the current implementation to control mplayer behaviour takes into account two options (solution for @tj-endpoint):

   - nothing, close - means that the video would play and after it's finished, the window would disappear. this is a simplification compared to the description above which had finer granularity of nothing - stay on the last frame of the video and close - close the window. Nothing, Close chosen in the CMS GUI would now lead to closed mplayer window. @tj-endpoint - is the 'stay on the last video frame' feature needed? if so - I'll investigate further

- loop / Restart the video - as was the previous behaviour - loop the video until the scene duration time.

The current solution (in appctl, lg_ros_nodes) implements respawn True/False flag which is propagated down the road to ProcRunner and breaks the while True loop based on the received activity_config value in the /director/scene message and accordingly translated into lg_media message. @wzin @mvollrath please review should you see something more appropriate than this.

the implementation is ready for review @wzin @mvollrath and I'll make a release for @tj-endpoint to be able to test and give feedback.

there are not tests associated yet, but there will definitely be.

zdenekmaxa, 2016-06-10:
this is implemented in
appctl 1.0.12
lg_ros_nodes 1.2.14
and available in this ISO:
https://iso.galaxy.endpoint.com/1.0.13-rc2-roscoe.iso
built here: http://lg-ci-01.endpoint.com:8080/job/ISO_build_test_and_publish/162/console

tj-endpoint, 2016-06-10:
I can't think of a case where we have wanted to leave the last frame up.  I think it's fine to set aside that condition, or wrap it into the close condition as you've mentioned, for now.  I would think it could add value to the CMS functionality in the future, however, and would pursue that option when time permits.  For now, it's most critical that movies can close when they are finished.

wzin, 2016-06-13:
@zdenekmaxa saw the appctl implementation - looks good. Since it may be a breaking change - we need to test all activities before rolling this out to production.

zdenekmaxa, 2016-06-15:
1.0.13-rc2-roscoe release is deployed on ephqlgxros, once you get a change @tj-endpoint please check the new functionality (close/nothing vs restart the videos) now works as desired and agreed above and was verified yesterday with @mababio:

FD:
```
Zdenek
so one plays just once and stops and the other restarts until the end of the scene, correct? 0:04
mababio
Yep
```

also, the UI needs the suggested changes to have always default and define only two options now (close/nothing and restart):

https://github.com/EndPointCorp/ros_cms/issues/447

and related:

https://github.com/EndPointCorp/ros_cms/issues/446

@ @eggyknap

tj-endpoint, 2016-06-16:
This is great @zdenekmaxa .  Thank you so much.  I have tested multiple videos in multiple scenes and they all indeed end now instead of looping.  The next step is to make this work for browser videos as well, since both types are spread throughout most of our presentations.

zdenekmaxa, 2016-06-17:
- rosnodes set up on portaldevcz so that browser video player windows pops up on the big display, but nothing is played in it
- investigated the commands that start chrome browser player, and found suspicious stuff, need to compare with a functional system where this feature works OK, expected feedback from @tj-endpoint on this
- otherwise, down the line, the browser video player is ManagedApplication so the implementation of the looping control will be analogous. didn't investigate yet the messages, analogy expected here too.

mikemerrill, 2016-06-17:
@zdenekmaxa Today while reviewing the content on devtn (where these changes have been recently implemented, I believe), we found that if the "when video completes" attribute is set to anything other than blank, the browser video player asset behaved erratically.
- it did not always appear, but sometimes did
- when there was a browser video player asset in the previously selected scene, that asset would persist, instead of the appropriate asset loading

zdenekmaxa, 2016-06-20:
@mikemerrill :

- nothing was done for browser video assets, everything so far, as confirmed above by @tj-endpoint is related solely to 'video' asset (which is mplayer played videos)
- scrutinized log files on devtn, nothing suspicious found
- are you 100% sure that the erratic behaviour started after setting up the 'when video completes' attribute? my guess is that the videos would behaves the same regardless of this setting modification (again, no implementation changes were done for this type of asset)
- what you describe above may be related to how scenes are queued, this would be best discussed among @eggyknap and @tj-endpoint and @b-berry and the  *exact* erratic behaviour scenario described. it's otherwise very difficult to reproduce it and distinguish between a bug and a feature
- i can now play browser videos on my system but there are some other issues apart from the lack of looping control which you perhaps hit here as well: (https://github.com/EndPointCorp/lg_ros_nodes/issues/236

I will say more in an email. Still, thanks a lot for testing the feature. For now, I'd like 'video' (mplayer) assets loop behaviour be tested in and out ( @tj-endpoint @jmerrill0376 )

wzin, 2016-06-20:
@mikemerrill @zdenekmaxa - web browser video player is something that we're not using too much anywhere - AFAIK mplayer (`video` type of asset) is used more frequently.

zdenekmaxa, 2016-06-20:
thanks for the comment @wzin but browser video player is something that 1) @tj-endpoint requested above and 2) with which I am finding more issues: https://github.com/EndPointCorp/lg_ros_nodes/issues/236

so the grave question is whether something rather unreliable like this browser player is worth the effort when we have mplayer.

zdenekmaxa, 2016-06-21:
- browser video player was identified as desired feature provided 'reasonable' amount of work

- first do https://github.com/EndPointCorp/lg_ros_nodes/issues/236

- if #236 is not entirely satisfactory, investigate browser video player alternatives

- implement loop control via respawn flag just as for mplayer videos

wzin, 2016-07-13:
This is working as expected. clsoing.

closed 2016-07-13 by wzin