EndPointCorp/lg_ros_nodes #195: use rosunit instead of nosetests

zdenekmaxa, 2016-04-08:
**Overview:**
nosetests runner misbehaves (ignores main section of the modules, ignores pytest directives - decorators, special methods). rosunit does the same job but behaves correcly.
ALL that is needed to is to replace nosetests command with rosunit in our tests_runner.py

opinions? @wzin @mvollrath @minshallj

wzin, 2016-04-11:
👍 for using anything but not nosetests. I remember many issues in Sept/Oct 2015 when you and @minshallj were working on tests.

zdenekmaxa, 2016-04-11:
that's right. more over,  you know how much biased I am, but really, the more I have to do with various testing frameworks, the more I like and want to migrate everything to pytest. Alas, this is not possible with catkin, but rosunit seems it will do the same job plus the benefits I am listing.
as far as catkin CMakefile is concerned, it recognizes just add_rostest, catkin_add_nosetests but in our custom runner which we need to have, we can execute whatever we like. So I am suggesting nosetests -> rosunit in the runner script.

mvollrath, 2016-04-12:
Why do we have a test_runner.py?  Catkin runs tests for us.  Here's the docs: http://docs.ros.org/indigo/api/catkin/html/howto/format2/index.html#configuring-and-running-unit-tests

minshallj, 2016-04-12:
remember that it wasn't properly returning error values when there were errors in the scripts themselves. `test_runner.py` just looks through all the CMakeList.txt files and grabs out all tests and puts them into nosetest and rostest run lists and then calls them manually.

zdenekmaxa, 2016-04-13:
@mvollrath it was so painful to rely fully on catkin tools. plus just as @minshallj said ...

closed 2016-04-14 by zdenekmaxa