EndPointCorp/lg_ros_nodes #199: fix lg_ros_nodes_development jenkins project

zdenekmaxa, 2016-04-14:
**Overview:**
the project started failing after spacenav merge (among all) from @minshallj 
it fails during catkin_make phase with

11:15:12 make[2]: *** [spacenav_wrapper/CMakeFiles/hid.dir/src/hid.c.o] Error 1
11:15:12 [  8%] make[1]: *** [spacenav_wrapper/CMakeFiles/hid.dir/all] Error 2
11:15:12 make[1]: *** Waiting for unfinished jobs....

11:15:12 [  1%] /var/lib/jenkins/jobs/lg_ros_nodes_development/workspace/catkin/src/spacenav_wrapper/src/hid.c:43:21: fatal error: libudev.h: No such file or directory
11:15:12  #include <libudev.h>
11:15:12                      ^
11:15:12 compilation terminated.

it runs fine on my local machine

**Testing scenario**
jenkins green again

zdenekmaxa, 2016-04-14:
it's missing dev library ...

lg_ros_nodes/catkin/src/spacenav_wrapper/src/hid.c:43:21: fatal error: libudev.h: No such file or directory
 #include <libudev.h>
                     ^
compilation terminated.

zdenekmaxa, 2016-04-14:
hm, althought the culprit's package.xml file contains this:
<build_depend>libudev-dev</build_depend>
(the requisite library), it's not handled by catkin

after manual installing libudev-dev, catkin_make compiles.

any explanation to this @wzin @mvollrath @minshallj  ?

minshallj, 2016-04-14:
ah, catkin_make is being run on jenkins itself, and I don't think we're installing dependencies manually via `rosdep`, we probably should.

zdenekmaxa, 2016-04-15:
it's been always doing catkin_make before running tests, both on development as well as on the master branch, which is followed by deb build

the latter was failing yesterday and it's fixed now. I'd like to understand why.

mvollrath, 2016-05-04:
Fixed this the other day.

closed 2016-05-04 by mvollrath