EndPointCorp/lg_ros_nodes #206: Prevent POV change while viewer is in background

mvollrath, 2016-05-04:
**Overview:**
When a Street View scene is loaded and then unloaded, the camera may continue to change while it's in the background and not be reset upon the next pano load.

Reported by @bengoldstein on `roscoe-1.8.0-rc2` on `ephqlgxros`.

**Requirements:**
List of requirements that need to be satisfied that are needed to consider this feature implemented.
- If a Street View asset specifies a pov, it must present that pov.
- If a Street View asset does not specify a pov, it must present a horizontal view.

mvollrath, 2016-05-09:
Since the street view handler always sets a fov, this was potentially due to an exception in the handler, potentially due to unexpected data from the CMS.

https://github.com/EndPointCorp/lg_ros_nodes/blob/e2f0999cb08f885dcef41aed50d4a6409c41f27e/lg_sv/scripts/server.py#L82-L115

wzin, 2016-06-03:
dupe of #561

mvollrath, 2016-06-09:
I think this was fixed in https://github.com/EndPointCorp/lg_ros_nodes/commit/f448a179691cc889e6105a16b78c9dc367468814 ?

mvollrath, 2016-06-14:
New scene data:

```json
{
      "activity": "streetview",
      "activity_config": {
        "panoid": "7lQbLQSjDnWlujR0uVyv6g"
      },
      "assets": [
        {
          "heading": null,
          "panoid": "7lQbLQSjDnWlujR0uVyv6g",
          "tilt": null
        }
      ],
      "height": 1920,
      "presentation_viewport": "left_three",
      "width": 1080,
      "x_coord": 0,
      "y_coord": 0
    }
```

mvollrath, 2016-06-14:
While testing today @BenWitten reported the view is still not horizontal when selecting a new pano from the map, but it appears to be horizontal when selecting a new presentation scene.

minshallj, 2016-06-24:
@mvollrath I'd guess that's because presentations will default to 0-heading and 0-tilt, but clicking on the touchscreen will just use whatever the previous POV is. Though the POV only changes while in streetview now (because of the commit mentioned above).

minshallj, 2016-06-24:
https://github.com/EndPointCorp/lg_ros_nodes/blob/master/lg_sv/scripts/server.py#L104-L117 That's where we set the POV when a scene with streetview assets is found by the streetview server.

closed 2016-07-08 by wzin