EndPointCorp/lg_ros_nodes #207: touching nav in streetview scene goes out to earth

minshallj, 2016-05-06:
```json
{"description": "rofl", "windows": [{"assets": [], "y_coord": 666, "x_coord": 666, "height": 666, "width": 666, "activity": "no_activity", "presentation_viewport": "left_two"}, {"assets": [], "y_coord": 666, "x_coord": 666, "height": 666, "width": 666, "activity"
: "no_activity", "presentation_viewport": "touchscreen"}, {"assets": [], "y_coord": 666, "x_coord": 666, "height": 666, "width": 666, "activity": "no_activity", "presentation_viewport": "center"}, {"assets": [], "y_coord": 666, "x_coord": 666, "height": 666, "width": 666,
 "activity": "no_activity", "presentation_viewport": "right_two"}, {"assets": [], "y_coord": 666, "x_coord": 666, "height": 666, "width": 666, "activity": "no_activity", "presentation_viewport": "left_one"}, {"assets": [], "y_coord": 666, "x_coord": 666, "height": 666, "w
idth": 666, "activity": "no_activity", "presentation_viewport": "right_one"}], "name": "Vader is watching", "duration": 666, "slug": "vader_watching", "resource_uri": "no uri"}
```

This is published by something (the touchscreen @axisofentropy ?) when touching the spacenav while in a streetview scene. It doesn't always happen, and I'm not sure what makes it happen. I would assume that it's trying to break the current presentation so the next scene doesn't autoplay.

minshallj, 2016-05-06:
Looks like the scene is coming from here: https://github.com/EndPointCorp/lg_ros_nodes/blob/36e8562515169d6279cadb4005b748819ebf7a39/lg_attract_loop/src/lg_attract_loop/attract_loop.py#L96

wzin, 2016-05-06:
@minshallj which system is this? I'm guessing it's tokyoros and here's what's happening:

- system has attract loop configured on it
- system is in SV 
- system becomes inactive (after 120s)
- someone breaks the inactivity by touching spacenav (or touchscren)
- system emits blank message because of `go_blank` default strategy for lg_attract_loop:
 - SV hides itself
 - there's no GE in the background

According to lg_attract_loop readme:

>~stop_action [string] - action that should be taken by LG after changing state from inactive to active:
go_blank - unload all the assets
load_presentation (experimental - supported only with director + cms)
stop_playtour - stop any tours that may have been being played back

we could emit `stop_playtour` message instead of emitting blank message. I think `load_presentation` is not implemented so we could prolly create some more strategies e.g.:
- show_earth
- play_welcome_message
- go_to_location_in_ge
- go_to_location_in_sv
- show_touchscreen_help

minshallj, 2016-05-06:
When this happened, the system was only inactive for  few seconds. I
noticed this on chm yesterday with Josh, but he mentioned that it was a
"known issue."

closed 2016-09-01 by wzin