EndPointCorp/lg_ros_nodes #220: Implement visibility callback and adhoc logical groups in AdhocBrowserPool

wzin, 2016-05-26:
**Overview:**

For multi-browser activities like e.g. streetview or sketchfab, we decided that these activities will preload in the background but won't show up on the screens until javascript applications running inside them (or an extensions that watches them) sends a signal to a ROS node that makes them visible. This ticket's requirements cover all work that needs to be done to support AdhocBrowserPool support for waiting and executing VISIBLE signal. NOTE: all browser's apps (js/extensions) will emit a signal that will be interpreted by an external ROS node that will emit these `/director/window/ready` signals that AdhocBrowserPool listens on.

**Requirements:**
- After spawning new `AdhocBrowser`, don't make it visible by default
- Make adhoc_browser_director_briwdge support more activities than only `browser` by exposing a ros parameter for setting a list of supported activities e.g. `streetview` (possibly) or `whatever`. In this way we want to create groups of content types inside AdhocBrowser that's going to become our main content delivery channel.
- Because we decided to have logical groups of content in AdhocBrowsers, add `activity_type` to [AdhocBrowser.msg](https://github.com/EndPointCorp/lg_ros_nodes/blob/master/lg_common/msg/AdhocBrowser.msg)
- In adhoc browser Pool - Listen on `/director/ready` - when a message containing `activity_type` is published on it - make all current browsers with this activity type VISIBLE.

mvollrath, 2016-05-26:
Keeping a list of supported activity types is expensive and should be avoided.

wzin, 2016-07-12:
https://github.com/EndPointCorp/lg_ros_nodes/commits/issue/164_activity_config

closed 2016-08-31 by wzin