EndPointCorp/lg_ros_nodes #223: Smooth transitions and preloading

wzin, 2016-05-26:
Overview:

A typical javascript application that runs inside of an adhoc browser needs few seconds before it's loaded.

An asset loaded into a browser

We need to show a splash screen, progress bar or just create a hook for showing intermediary image before application is fully loaded and can be shown on the screen. This calls for a hook sent from application to the adhoc browser pool's browser instance (that js app runs in) for delayed ApplicationState.VISIBLE signal. Each javascript application would have to have know the unique ID of browser instance it's running inside of. In order to do this we'll need to append sth like ros_instance_id to get arguments of the browser so an extension or the application itself can send ApplicationSate.VISIBLE to the browser instance.

Design graph: https://drive.google.com/file/d/0B1VEo7XUZNtwd29lNGJfbXFRaE0/view?usp=sharing

wzin, 2016-05-26:
@minshallj asked about prevent audio from being played back when the app is being preloaded in the background.

wzin, 2016-05-26:
@mvollrath suggests using window API for chrome: https://developer.chrome.com/extensions/windows 
https://www.flowdock.com/app/endpoint/r-d/threads/QvbXcK2UoR1kNGwCg6RYVC8BSeP

kiselev-dv, 2016-08-25:
We have some web applications which are not allows us to use custom get arguments, like google maps.

And we need some way to pass arguments from command line args from managed application into the chrome extension. As an Idea, we can use user agent string. It contains some sections, one of the sections is the section about host system. So we can add there some our stuff, like ros_node_name and such stuff. UA is easy accessible from js, and do not requires us to make a call for json data on the host.

closed 2016-08-31 by wzin

wzin, 2016-09-05:
@kiselev-dv another way to do that is to generate a static small html web page that would load first and redirect to google maps from it.