EndPointCorp/lg_ros_nodes #230: AttributeError: PanoViewerServer instance has no attribute '_initialize_variables'

zdenekmaxa, 2016-06-15:
I am seeing following Python errors on ephqlgxros:

```
 ^[[0m^[[31m[ERROR] [WallTime: 1465818934.261193] bad callback: <function cb at 0x7f106e931c08>
 694 Traceback (most recent call last):
 695   File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/topics.py", line 720, in _invoke_callback
 696     cb(msg)
 697   File "/opt/ros/indigo/lib/python2.7/dist-packages/lg_common/helpers.py", line 623, in cb
 698     func(msg)
 699   File "/opt/ros/indigo/lib/python2.7/dist-packages/lg_sv/server.py", line 363, in _handle_soft_relaunch
 700     self._initialize_variables()
```

```
  4 Traceback (most recent call last):
  5   File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/topics.py", line 720, in _invoke_callback
  6     cb(msg)
  7   File "/opt/ros/indigo/lib/python2.7/dist-packages/lg_common/helpers.py", line 623, in cb
  8     func(msg)
  9   File "/opt/ros/indigo/lib/python2.7/dist-packages/lg_sv/server.py", line 363, in _handle_soft_relaunch
 10     self._initialize_variables()
 11 AttributeError: PanoViewerServer instance has no attribute '_initialize_variables'
```

_initialize_variables is called from lg_sv/src/lg_sv/server.py but I somehow fail to see this methods's definition - which would explain the AttributeError really. Or am I missing some higher python metaclassing overdynamic school here, @mvollrath ?

mvollrath, 2016-06-15:
It looks like it's calling a method that doesn't exist.

minshallj, 2016-06-15:
I think that method used to exist, I remember making it once upon a time

minshallj, 2016-06-15:
ok, so it was taken out of the __init__ method, but the call to _initialize_variables was left in the _handle_soft_relaunch function. @mvollrath it looks like you removed it 49d52e71778a65e3ed4a6aecf775f794f0e4e31c here. Did you do that to simplify the init? The reason I put that in there was to reset the state of the program when we wanted to soft relaunch, so the current pov and panos were cleared out.

closed 2016-06-15 by mvollrath

zdenekmaxa, 2016-06-16:
yes, test. always, test would have told immediately.
nice commit @mvollrath

commit 48be96e7 (2016-06-15): Fix #230 and add tests