EndPointCorp/lg_ros_nodes #236: browser video player asset sometimes not played

zdenekmaxa, 2016-06-20:
some video files have problems to play in the chrome browser player:

[1:1:0617/092930:ERROR:render_media_log.cc(23)] MediaEvent: PIPELINE_ERROR demuxer: could not open

this is one of the issues caught, it seems it's a chromium problem.

when this happens, the browser window appears on the big display but nothing is played.

interestingly, when this happens, the command which attempts to start the browser video player looks like this:

/usr/bin/google-chrome --remote-debugging-port=34818 --user-data-dir=/tmp/lg_browser-adhoc_media_browser_display_wall_portal --disk-cache-dir=/tmp/lg_browser-adhoc_media_browser_display_wall_portal --crash-dumps-dir=/tmp/lg_browser-adhoc_media_browser_display_wall_portal/crashes --no-first-run --allow-file-access-from-files --disable-default-apps --disable-java --disable-session-storage --disable-translate --touch-events=enabled --disable-pinch --overscroll-history-navigation=0 --disable-touch-editing --log-level=0 --enable-logging --v=1 --enable-webgl --ignore-gpu-blacklist --kiosk http://localhost:8008/lg_sv/webapps/videosync/index.html?src=http://lg-head:8088/roscoe_assets/\&syncRate=60\&maxPlaybackRate=1.5\&frameLatency=0\&hardSyncDiff=1.0\&minPlaybackRate=0.5\&rosbridge_port=9090\&pingInterval=1000\&show_controls=false\&leader=true\&rosbridge_host=localhost\&autoplay=true

-> it clearly misses the video file in the source URL. 

this has been caught several times on portaldevcz with avi video files.

a few mp4 tried as browser video player assets are played OK and the URL does contain the correct file name.

- the video file format issues should be investigated - how much can we generally rely on browser video player?

- the missing file name should be investigated, very misleading, to say the least

comemnts please: @wzin @mvollrath

zdenekmaxa, 2016-06-20:
one more issues with the browser player - on a video file (mp4) which actually plays, the above command loops the file indefinitely. when inside a CMS scene, it would only end after a scene duration time is over. implementing any higher-lever close/loop control like in #193 is futile since the player app itself loops indefinitely. if the looping behaviour in the browser video player is pursuit (despite the format/codec issues mentioned above), the behaviour of the player itself should first be put under control.

@wzin - was this ^^ known? either of the issues mentioned here?

zdenekmaxa, 2016-06-20:
although a video file is played indefinitely (until the end of the scene) it -sometimes- disappears from the big display. when this happens, the chrome process still hangs there, the chrome debug log file /tmp has lots of error statements like 

861 [29062:29062:0620/122216:INFO:CONSOLE(1)] "WebSocket connection to 'wss://42-b:9090/' failed: Error in connection establishment: net::ERR_CONNECTION_REFUSED", source:     chrome-extension://ncekbjgokljfdjegpcfpehjfihibekoe/libs/roslib.min.js (1)
862 [29062:29062:0620/122216:INFO:CONSOLE(221)] "AlbatRos caught an error: ", source: chrome-extension://ncekbjgokljfdjegpcfpehjfihibekoe/libs/albatros.js (221)
863 [29062:29062:0620/122216:INFO:CONSOLE(212)] "AlbatRos connection closed", source: chrome-extension://ncekbjgokljfdjegpcfpehjfihibekoe/libs/albatros.js (212)

upon a second attempt to enter the scene and it's video, the video window is just blank.

zdenekmaxa, 2016-06-21:
- browser video player was identified as desired feature provided 'reasonable' amount of work

- first address the player default looping

- won't probably be able to do much about codec issues

wzin, 2016-07-13:
Some scenes have broken location for the asset so

>[1:1:0617/092930:ERROR:render_media_log.cc(23)] MediaEvent: PIPELINE_ERROR demuxer: could not open

it's an error saying "could not open the file" because url to the source file, provided in `src` get arguemtn is following:

`http://lg-head:8088/roscoe_assets/`

e.g. broken

closing

closed 2016-07-13 by wzin