EndPointCorp/lg_ros_nodes #237: report ros nodes test coverage

zdenekmaxa, 2016-06-21:
**Overview:**

- investigate tools to map lg_ros_nodes test coverage (start with py.test cov plugin), the tool needs to play nice with ROS if necessary (seems like probably not, just static analysis of the codes vs test cases should suffice). still, check what people do in the ROS universe

- brief manual review of no coverage - file a ticket per node where necessary

- the coverage report tool shall be pluggable into CI and the report send to the team members

- bonus points will be checking on push / commit basis and warnings (or even build failures on uncovered code)

zdenekmaxa, 2016-08-31:
The requirements were a little too ambitious here.

While all of the above points are virtually achievable, arriving at grand summary pairs: 

"ROS_NODE_NAME ... per module REAL coverage percentage"

would be very hard. The main obstacle here are online and offline test cases variants. While getting PRECISE test coverage based on offline tests (per module granularity, reporting lines with no test coverage) is easy, combining this information with coverage from online tests would be very difficult. Online tests are for example instantiating ROS node classes automatically, just by nature of running rostest testdef.test file. This fact is not obvious to the coverage report tool and the constructor is reported as missing from the tests, which is not true. Another example would be message callbacks. For ROS nodes with both online and offline tests, we'd need to check online reports and offline reports, the latter being not PRECISE. Rather than having 1 result.

As far as I was able to search, there is nothing ROS-native, the recommendation is 

```
rostest.rosrun(package, test_name, test, sysargs="--cov")
```

which works OK. The results are produced in the log files and are very detailed:

```
Name                                         Stmts   Miss  Cover
----------------------------------------------------------------
lg_common/interactive_messages_factory.py       59     59     0%
lg_ros_nodes/lg_common/src/lg_common/interactive_messages_factory.py:
Missing lines: 1-529
```

While we would prefer to have the the coverage report in the main console log output, it is put into a separate log file with cryptic ROS file name.

Also, the rostest tool (which on lower layer just calls standard unittest) does not consider coverage implemented by the other unittesting framework we use - the pytest. It is not the case the other way round, quite expectantly.

py.test coverage used in the ROS offline tests works OK. On the online tests, the arguments which trigger the pytest-cov are somehow ignored, coverage wasn't generated.

The pytest-cov report looks like this:

```
----------------------------------------------------------------- coverage: platform linux2, python 2.7.6-final-0 ------------------------------------------------------------------
Name                          Stmts   Miss  Cover   Missing
-----------------------------------------------------------
setup.py                          4      4     0%   3-13
src/lg_offliner/offliner.py     132    122     8%   2-35, 47, 53, 70-236
```

This again shows the discrepancy between online and offline tests. While src/lg_offliner/offliner.py is very thoroughly tested, the coverage is reported only based on offline tests, the online tests are not taken into account.

As stated above, this works OK with rosunit tool (offline tests):

```
pytest.main("-vv -rfsX -s --junit-xml=%s --cov=. --cov-report term-missing %s" % (pytest_result_path, test_path))
```
but not on via rostest tool (online tests), no coverage data is generated.

Again, the coverage output is put into a separate ROS log file. pytest options to output everything into std output are not working.

The conclusion is to manually review our ROS nodes and report nodes with no coverage at all and implement at least SOME tests there.

Producing sensible coverage percentage would be very difficult for the above reasons.

wzin, 2017-01-31:
Maybe we could use coveralls.io?

closed 2019-07-09 by mvollrath