EndPointCorp/lg_ros_nodes #238: director scene tracker

minshallj, 2016-06-22:
**Overview:**
Sometimes it is nice to be able to query the last scene (or last few scenes) sent over the /director/scene topic.  Maybe we should make a general ros node that can track different topics, and return the last N messages that were sent out over that topic. Or you pass it a number, 1 for the last message, 2 for the one before that... all the way up to N and it'll return just that message. This can be useful for streetview as well.

**Requirements:**
List of requirements that need to be satisfied that are needed to consider this feature implemented.
- Track topics by scene
- Store the last N
- Make a service call that returns the nth message

**Development notes**
Not sure if this should be a stand alone node, or just a class we import in each script. Importing might be easier since we won't need to do anything tricky to import the proper topic type dynamically.

**Testing scenario**
See if it works :]

axisofentropy, 2016-06-23:
What are some use cases or user stories for this?

minshallj, 2016-06-23:
It's more for development / debugging. I'd like to jump on a system and be able to see what state it's in without having to wait around listening on multiple topics for a new state to change

axisofentropy, 2016-06-23:
ok that's good info.  If it only needs visibility to us support peoples,
then it could just be a log file or three.  I bet ROS has some feature
for logging what flies across topics.

wzin, 2016-06-23:
There are at least two very much related tickets:
- https://github.com/EndPointCorp/lg_ros_nodes/issues/168 - being able to
relaunch to last state (also relaunch ros nodes selectively)
- https://github.com/EndPointCorp/lg_ros_nodes/issues/165

mvollrath, 2019-07-09:
There is now a service to query the current scene.

closed 2019-07-09 by mvollrath