EndPointCorp/lg_ros_nodes #273: Make it possible to make adhoc browser window always on top

wzin, 2016-09-02:
**Overview:**

Use case: mirroring with small browser instance on the touchscreen for controls.

**Requirements:**
- Make it possible to set window ontop'ness in uscs's message activity config
```json
{
"activity_config": 
   {"window":{
      "ontop": true
    }
   }
}
```
- it should make its way down to awesome rules https://github.com/EndPointCorp/lg_ros_nodes/blob/f3b48c1bab815f8886a2a31eb88d74c7de2fee59/lg_common/src/lg_common/awesome.py#L84
- make it an AdhocBrowser.msg bool flag

mvollrath, 2017-01-23:
I experimented with this recently and the results were not promising.  Setting `above` `below` or `ontop` in properties list and/or callback did not seem to affect the relative positioning of Chrome windows.  It may be a problem specific to Chrome or our hack to run it in `--kiosk` mode with window positioning (it always wants to be fullscreen).

mvollrath, 2017-03-04:
This needs to be looked at again.

wzin, 2017-03-08:
#1314 is prolly related