EndPointCorp/lg_ros_nodes #287: Stuttering during flight

mattjg908, 2016-09-26:
@BenWitten commented on [Mon Sep 26 2016](https://github.com/EndPointCorp/lg_touchscreens/issues/158)

**Overview:**
Occasionally while flying to a location, the system will stutter while en route. The stuttering consists of smooth flight followed by a sudden stop and then continuing in flight. This is an issue that should not happen. 

**Requirements:**
-If this stuttering issue does not continue to happen, then this issue can be resolved and closed.

**Testing scenario**
-Fly to locations consistently without this stutter happening. 

Two examples of this issue can be found below:
1. https://drive.google.com/file/d/0B_uLPLWkN3AoeGJidHVVc2pScEE/view?usp=sharing
2. https://drive.google.com/file/d/0B_uLPLWkN3AoMVpWVG5hbHNpZkk/view?usp=sharing

BenWitten, 2016-09-26:
Per @mattjg908 's request, I have replicated this issue. It can be seen here: https://drive.google.com/file/d/0B_uLPLWkN3AoMFAyeEZtSTY1dm8/view?usp=sharing

BenWitten, 2016-10-14:
This should be prioritized. It happens every time the user moves on the tablet/touchscreen.