EndPointCorp/lg_ros_nodes #291: Panoramic Video Functionality poor when asset_type Panoramic Image Viewer

eggyknap, 2016-09-28:
@b-berry commented on [Wed Sep 28 2016](https://github.com/EndPointCorp/ros_cms/issues/583)

**Overview:**
Clients are increasingly requesting panoramic video content ie 360 video.  The current Panoramic Image Viewer node supports very limited *.mp4 video playback.  

**Requirements:**
- Support 360 Video Playback and nav interaction of assets like: `wget https://www.dropbox.com/s/8tjnso8imu4gfax/Tubbataha_360.mp4?dl=1`

```
(ephqlgxros)galadmin@lg-head-337:/media/lgmedia/ros_cms_default_assets
13:41:21$ avprobe Tubbataha_360.mp4 
avprobe version 9.18-6:9.18-0ubuntu0.14.04.1, Copyright (c) 2007-2014 the Libav developers
  built on Mar 16 2015 13:19:10 with gcc 4.8 (Ubuntu 4.8.2-19ubuntu1)
Input #0, mov,mp4,m4a,3gp,3g2,mj2, from 'Tubbataha_360.mp4':
  Metadata:
    major_brand     : mp42
    minor_version   : 0
    compatible_brands: mp42mp41
    creation_time   : 2016-07-08 03:38:07
  Duration: 00:05:58.72, start: 0.000000, bitrate: 45391 kb/s
    Stream #0.0(eng): Video: h264 (Main), yuv420p, 3840x1920, 45067 kb/s, 60 fps, 60 tbr, 60k tbn, 120 tbc
    Metadata:
      creation_time   : 2016-07-08 03:38:07
    Stream #0.1(eng): Audio: aac, 48000 Hz, stereo, fltp, 317 kb/s
    Metadata:
      creation_time   : 2016-07-08 03:38:07
# avprobe output
```

**Testing:**
- Test scene w/ this asset as Panoramic Image Viewer type current located here:
- https://ephqlgxros.galaxy.endpoint.com/#/group/10/presentation/49/scene/679

mvollrath, 2017-03-07:
Why would we overload the "Panoramic Image Viewer" asset type with video playback?

eggyknap, 2017-03-09:
Only because it's the only option readily apparent in the CMS currently, I imagine, but @b-berry opened the original issue so he may have a better answer. I see rumors in various chat logs about a panoramic video player, which is sweet. If the CMS needs any special love to support it, its issue tracker has yet to hear about it.

mvollrath, 2017-03-15:
Thanks @eggyknap, writing something up since I'm working on backend for it.