EndPointCorp/lg_ros_nodes #298: Ad hoc browser in initial state does not always launch

mvollrath, 2016-10-05:
**Overview:**
I observed intermittent missing browsers while running this initial state:

```json
{
  "description": "janus mirroring scene",
  "duration": 0,
  "name": "janus_mirroring",
  "resource_uri": "bogus",
  "slug": "janus_mirroring_stock_message",
  "windows": [
   {
      "activity": "browser",
      "activity_config": {
        "preload": true
      },
      "assets": [
        "http://lg-head/mirror_playback_kiosk-loader.html"
      ],
      "presentation_viewport": "kiosk",
      "width": 1920,
      "height": 1080,
      "x_coord": 0,
      "y_coord": 0
    },
    {
      "activity": "browser",
      "activity_config": {
        "preload": true,
        "google_chrome": {
          "additional_cmd_args": [
            "--force-device-scale-factor=2"
          ]
        }
      },
      "assets": [
        "http://youtube.com"
      ],
      "presentation_viewport": "wall",
      "width": 3840,
      "height": 2160,
      "x_coord": 0,
      "y_coord": 0
    },
    {
      "activity": "mirror",
      "activity_config": {
        "route_touch": true,
        "viewport": "wall"
      },
      "assets": [
      ],
      "presentation_viewport": "kiosk",
      "width": 1920,
      "height": 1080,
      "x_coord": 0,
      "y_coord": 0
    }
  ]
}
```

I saw a missing kiosk and a missing wall.

**Requirements:**
* Browsers always launch when specified in `initial_state.json`.

**Testing scenario**
* .test file with initial state rig pointed to scene with a bunch of ad hoc browsers (maybe 50?)
* Assert that each browser exists
* Use Xvfb

wzin, 2016-10-13:
Made a simple test for that https://github.com/EndPointCorp/lg_ros_nodes/blob/master/scripts/relaunch_test.sh

wzin, 2016-10-17:
Made a lot of refactoring to make adhoc browser pool wait for connections on all topics that it subscribes and publishes to. Despite tht fact that connection count is not the best way of checking if all nodes are up, it waits for the most crucial readiness node to come up in time so this solution is working.

mvollrath, 2016-10-17:
Works consistently in 1.4.18

closed 2016-10-17 by mvollrath