EndPointCorp/lg_ros_nodes #321: Streetview should wait 5 seconds before readjusting view to be level

BenWitten, 2017-01-10:
**Overview:**
Currently, when you look up or down on streetview, the system waits about 2 seconds before evening its view. This is too fast, and as far as I can tell there is no reason the adjustment would have to happen so quickly. 

**Requirements:**
-Please set the system to wait 5 seconds before re-adjusting view on SV to be even.

**Testing scenario**
-Go to Streetview scene. Look up. Stop touching joystick. System should wait 5 seconds before adjusting view to be level. If system waits 5 seconds, this is a success.

wzin, 2017-01-13:
@BenWitten can you please consult @mvollrath who implemented this as well ass @bengoldstein and @rphipps who will have a lot of input on that?

This change will require fleet wide ISO deployment which is needed for some time now.

BenWitten, 2017-01-19:
@rphipps @mvollrath @bengoldstein what was the rationale behind determining how long before streetview leveled? It would be great to have a quick conversation about this.

bengoldstein, 2017-01-19:
@BenWitten, its current speed was just selected empirically back when. Matt configured this and we liked it. There may have been some futzing back and forth with the speed of the re-leveling that I weighed in on, although I don't specifically recall that.  I am fine with slowing it down to 5 seconds as you are suggesting.

rphipps, 2017-01-19:
Can we hear what the use case is for 5 seconds?

We tried a longer time period and it didn't feel as great when "walking 
down the path" in SV.

mvollrath, 2017-01-20:
* Snap timeout is [hardcoded](https://github.com/EndPointCorp/lg_ros_nodes/blob/64f4c8b141a41a2cdc6903bde377a024f8e09f87/lg_sv/src/lg_sv/server.py#L32) and needs to be parameterized.
* Let's creep up the timeout until it feels right to @BenWitten

BenWitten, 2017-01-20:
Thanks @mvollrath happy to work with you on this whenver is convenient.

FuriousJulius, 2018-12-06:
lg_ros_nodes 1.18.22 has a ros_param to set this time.

FuriousJulius, 2018-12-06:
https://github.com/EndPointCorp/lg_ros_nodes/commit/fba01ad02bb5cda8fb6d82a7b156c3926cb8a596

closed 2018-12-06 by FuriousJulius