EndPointCorp/lg_ros_nodes #322: getting lg_sv  production ready

minshallj, 2017-01-10:
The first issue is viewsyncing for tilt. I believe @mvollrath might know how to go about this. He mentioned using some sort of magic to get the screens to viewsync. We'd like to have this.

We also need to investigate zoom levels. There was a time when we had to abandon this because we could not zoom out far enough.

Also I remember performance being semi poor on this. Lets investigate that also.

mvollrath, 2017-01-10:
The trick we used for another application to add roll was to build a scale+rotation matrix in JS and send it to CSS transform.  Here's a snippet illustrating how this works:

```
var buildTransformMatrix = function(scale, roll) {

    // Scale x and y by same amount.
    // https://en.wikipedia.org/wiki/Scaling_(geometry)#Matrix_representation
    var s = scale;
    var scaleMatrix = [
        [s, 0, 0, 0],
        [0, s, 0, 0],
        [0, 0, 1, 0],
        [0, 0, 0, 1]
    ];

    // Rotate around z.
    // https://en.wikipedia.org/wiki/Rotation_matrix#Basic_rotations
    var sr = Math.sin(roll);
    var cr = Math.cos(roll);
    var rotationMatrix = [
        [cr, -sr, 0, 0],
        [sr, cr, 0, 0],
        [0, 0, 1, 0],
        [0, 0, 0, 1]
    ];

    // Multiply the matrices.
    var finalMatrix = matrixMul(rotationMatrix, scaleMatrix);

    // Flatten the matrix to an array.
    var flatMatrix = Array.prototype.concat.apply([], finalMatrix);

    // Produce the CSS transform string.
    return ['matrix3d(', flatMatrix.join(), ')'].join(' ');
};

var container = document.getElementById('someDiv');
container.style.transform = buildTransformMatrix(myScale, myRollInRadians);
```

In practice the scale should be some fixed value (2.0?) and the camera fov divided by that value to compensate.  You can save some trivial JS time/garbage by pre-generating the scale matrix and multiplying your dynamic rotation matrix into it.

This `matrix3d` approach turned out to be a lot faster than using `scale3d` combined with `rotate3d`.

mvollrath, 2017-02-17:
Moving ahead with a default oversize factor of 2.04, which should allow full tilt at 16:9.

We need to set [`window.devicePixelRatio`](https://developer.mozilla.org/en-US/docs/Web/API/Window/devicePixelRatio) equal to the oversize factor to preserve image quality.

Field of view/zoom formulae:

```
zoom = -log2(tan(PI * hFov / 360)) + 1
hFov = atan(2^(1 - zoom)) * 360 / PI
```

The zoom level from hFov works perfectly without oversizing, but a magic fudge factor needs to be applied to the input when dealing with an oversized canvas.  This is solved for an oversize factor of 2.04 in this implementation.

There are (new?) limits on horizontal field of view in the API.  When the window is square or landscape, the max horizontal field of view is around 60 degrees.  It goes much wider when the window is vertical.  This matters mostly because with an oversize factor, we hit this limit at roughly half the visible field of view.

Known working hFov when the window is 16:9 portrait: [12, 30]
Known workin hFov when the window is 16:9 landscape: [15, 63]

If higher values are called for by an installation, this can be worked around by:

* Reducing the oversize factor *and* maximum tilt (and figuring out the right oversize zoom fudge factor).
* Reducing (fudging) the field of view uniformly, thus breaking the cylindrical projection.

Oversizing the canvas obscures some essential elements:

* Pano data / attribution at the bottom.
* Link chevrons at least partially hidden.
* The Google logo in the bottom left corner.

Data and attribution have been replaced with simple text attribution.  Others yet to come.

mvollrath, 2017-02-20:
Link chevrons were implemented by covering the page in a THREE.js canvas with transparent background.

mvollrath, 2017-02-20:
Added the Google logo back.

mvollrath, 2017-02-21:
There is a [working roslaunch config](https://github.com/EndPointCorp/lg_ros_nodes/blob/master/lg_sv/launch/sv.launch) for testing.  Run that config and [publish a Street View scene with the test script](https://github.com/EndPointCorp/lg_ros_nodes/blob/issue/322_streetview/lg_sv/test/test_scene.py).

wzin, 2017-02-28:
Issues observed today with rc18
- first pano never loads (black screen + google logo on the bottom)
- loading non-first pano sometimes results in a short flicker with SV map on it
- performance with multi-yaw setup is the same on GF 450 as previously

wzin, 2017-03-09:
To retrieve metadata for new panoids, we could use streetview panoramic data
```js
  var width = panoData.Data.image_width;
  var height = panoData.Data.image_height;
  var tileSize = panoData.Data.tile_width;
  var maxZoom = panoData.Location.zoomLevels;
  var panoId = panoData.Location.panoId;
  var panoYaw = panoData.Projection.pano_yaw_deg;
```

wzin, 2017-03-09:
`https://www.google.nl/maps/photometa/v1?pb=!1m2!1smaps_sv.tactile!11m0!2m2!1sen!2snl!3m3!1m2!1e3!2s-{}!4m19!1e1!1e2!1e3!1e4!1e5!1e6!1e8!2m1!1e1!4m1!1i48!5m1!1e1!5m1!1e2!6m1!1e1!6m1!1e2`

just replace the `{}` with new style panoid

wzin, 2017-03-21:
Estimation for this ticket is only about fixing lg_sv_nonfree

wzin, 2017-03-22:
Some photospheres are black and show warning sign. Some of them are not stiched properly.

warning sign:
- `-EHSZMhePkF4/VdOf8qWPIlI/AAAAAAAAAFk/us7ur-ou6JwaZx1xTo8KBTbauCroYURuwCJkC`
- `-Wva53H-TpuQ/Utxqi30k3zI/AAAAAAAA6RE/RZMvgmnuo10ds8t5hE7lESMW1_9IysoWwCJkC`

improper stitching:
- `-g13kt2ekmi8/Vi2AaAydWgI/AAAAAAAAIZY/Md1XU4d_2echRCFXuSf0RQCO4MKDS-FxACJkC`
- `-s6VUZmDPdoo/WA4sKcae_FI/AAAAAAAAu9c/obFlQIfF0WEr1JAhwI_iHKezfhAw6J0sgCLIB`

Both issues are caused by improper metadata. Eg. 

https://lh3.ggpht.com/-g13kt2ekmi8/Vi2AaAydWgI/AAAAAAAAIZY/Md1XU4d_2echRCFXuSf0RQCO4MKDS-FxACJkC/x0-y0-z5/p

^ this returns warning sign

https://lh3.ggpht.com/-g13kt2ekmi8/Vi2AaAydWgI/AAAAAAAAIZY/Md1XU4d_2echRCFXuSf0RQCO4MKDS-FxACJkC/x0-y0-z1/p

but this returns zoomlevel = 1 of the image

We will need to run some additional algo for stitching here.

closed 2017-04-10 by wzin