EndPointCorp/lg_ros_nodes #330: Panoramic Video Player demo

mvollrath, 2017-02-07:
A panoramic video player webapp has been tacked on to lg_sv for demonstration.

It lives at `lg_sv/webapps/panovideosync/` and has some options:

* `videoUrl`: required, url to the *equirectangular* panoramic video.  [compatibility](https://developer.mozilla.org/en-US/docs/Web/HTML/Supported_media_formats)
* `hFov`: horizontal FOV in degrees.
* `yawOffsets`: comma-separated list of yaw offsets in degrees.
* `master`: if true, this is the instance that acts as the clock master and plays audio.  Must have one master.

For SpaceNav, run an lg_sv `server.py` instance with `server_type` param set to `panovideo`.

To add scenes, make a `Web Page` asset for each display node and span all main displays with it and configure the video, fov and offsets.  The master should be whichever you want audio playing from.

Upload videos to the head node `/var/www/`.  Note that [the server must support byte serving](http://superuser.com/questions/451445/how-can-i-test-if-a-server-supports-byte-serving/451447) and all head nodes do at this path.

joshuanorman, 2017-02-17:
Unfortunately still unable to check on qatn with iso 2.2.0-rc11 -- webgl seems to broken

wzin, 2017-02-28:
Closing - it's demoable.

closed 2017-02-28 by wzin