EndPointCorp/lg_ros_nodes #333: Make support for trusting self-signed certificates

wzin, 2017-02-20:
**Overview:**

Consider following scenario:
- browser pool is configured to trust an SSL certificate that's available under `http://lg-head:8080/self-signed.crt`
- when pool starts up, it automatically trusts the cert by adding it to system's `nssdb`:

```bash
certutil -d sql:${HOME}/.pki/nssdb -A -t "P,," -n self-signed-crt-host -i self-signed-crt-host
```

**Requirements:**
- make adhoc browser pool attempt to fetch all certificates that are mentioned in ros params that are starting with `~ssl_trust_*`
- make adhoc browser pool iterate over all URLs that are values to abovementioned rosparams and execute addition/trust command idempotently:
```bash
certutil -d sql:${HOME}/.pki/nssdb -A -t "P,," -n self-signed-crt-host -i self-signed-crt-host
```
- make the pool crash after 10 attempts of not being able to download and trust the certs
- make sure that nothing interferes with `nssdb` between relaunches

minshallj, 2017-04-05:
we need to not start any browser before all certificates have been trusted

minshallj, 2017-04-05:
So we followed instructions on how to make a https server & self signed key here:

https://anvileight.uk/blog/2016/03/20/simple-http-server-with-python/

But even after running the certutil command manually, we still get the chrome "not trusted" page

axisofentropy, 2017-04-12:
Can we add to Chrome's trusted CA list instead of one individual certificate? For example, we could configure LG Chrome to trust End Point's existing CA, then use it to easily sign lots of certs.

jonjensen, 2017-04-14:
@axisofentropy We definitely shouldn't be using literal self-signed certs, as you surmise. But End Point's existing CA is end of life. We're making a new one, but we wouldn't use that for LG stuff anyway. The LG infrastructure should have its own CA, I think.