EndPointCorp/lg_ros_nodes #349: Make support for default GET arguments in AdhocBrowserPool

wzin, 2017-04-03:
**Overview:**

This ticket is the other half of the work that's speced out here: https://github.com/EndPointCorp/lg_chef/issues/1273

**Requirements:**
- Make it possible to provide list of GET arguments that will be passed to each and every instance of the AdhocBrowser
- make it be a param

minshallj, 2017-04-03:
arguments should make their way into `/var/www/lg/lg-head.xml` under `/global_get_args/*` or something like that. Then adhocBrowserPool should grab all key/values and append `<url>?<key1>=<val1>&<key2>=<val2>` etc

minshallj, 2017-04-03:
test w/ tactile first please to make sure this is even possible.

axisofentropy, 2017-04-24:
Another use case for this: Right now when launching Gigapixel, GET parameters for the `portal-launcher` Quick Menu don't get passed into the URL. Other applications use an intermediate "launcher" HTML page, but Gigapixel does not (and probably should not) so we need another way to get those passed.