EndPointCorp/lg_ros_nodes #357: Copy NaCl Chrome component into Managed Browsers

axisofentropy, 2017-05-03:
**Overview:**
Chrome generally installs a PNaCl component when it first runs. Because our managed Chrome browsers each use an ephemeral profile directory, they don't have an opportunity to do this before we need them to perform. Our experience with the Flash component suggests we can install this component ourselves just before starting the Chrome process.

**Requirements:**
- Managed Browsers have the PNaCl component installed immediately upon launch.
- NaCl content, specifically New Earth, functions consistently even after reboot.

**Development notes**
Flash is installed within `lg_common/src/lg_common/managed_browser.py`, so try imitating that implementation.

**Testing scenario**
- New Earth launches within a Managed Browser immediately after a reboot.  (Otherwise just a flat black browser.)

bengoldstein, 2017-08-07:
Has this been tested? If so, let's get it out of the testing column. If not, not let's get it tested and get it out of the testing column. :-)

axisofentropy, 2017-10-09:
Yeah I'm pretty confident of this.

closed 2017-10-09 by axisofentropy