EndPointCorp/lg_ros_nodes #366: Provide current Earth pose for CMS Scene Editor

axisofentropy, 2017-05-18:
**Overview:**
Our real estate client and others desire faster Google Earth bookmark development by creating bookmarks from their running LG system.  This use case is described in detail in the associated `ros_cms` ticket https://github.com/EndPointCorp/ros_cms/issues/377

We'd like to develop this within the Scene Editor, so that web interface requires access to Google Earth's most recent pose.

**Requirements:**
- Maintain state: most recent Earth pose published on topic
- Provide state: in response to requests

**Development notes**
A stateless HTTP service is preferred, tho we can probably connect to a ROS service if HTTP is not feasible.

**Testing scenario**
Unit test:
* Make request to service.
* Verify valid response reflects Google Earth's position.
* Repeat in a few different system states: just after reboot/relaunch, Earth in background, etc.

Integration test: Works with the implementation described in https://github.com/EndPointCorp/ros_cms/issues/377

eggyknap, 2017-06-09:
It has been pointed out, correctly, that this becomes much easier if we assume the CMS in question runs on the LG's head node, and that getting LG coordinate data to some CMS running elsewhere would be very hard. Personally I'm very happy saying "this will only work when the CMS is running on the LG's head node".

eggyknap, 2017-06-09:
We can get started on this by having the cms just run `rostopic echo /earth/pose` in the backend upon request to some API endpoint.

minshallj, 2018-04-06:
@eggyknap has this been done?

eggyknap, 2018-04-06:
Yes.

closed 2018-04-06 by eggyknap