EndPointCorp/lg_ros_nodes #377: Improve sync_to_displaynodes

FuriousJulius, 2017-10-04:
**Overview:**
Lets give the `sync_to_displaynodes` script an option to update our checkouts (that are already on master) of the various repos in the systems `catkin/src/`.  This will help ensure you aren't testing outdated combinations of code.  Recently Dmitry went to test his cesium changes. The lg_cesium repo was on the branch he wanted to test, lg_ros_nodes repo was up to date but appctl was a single commit too old for everything to work.  Instead of having him manually fetch/pull each branch in his catkin/src lets do this automatically warning/ignoring if a repo is on something other than master.

**Requirements:**
-Safely pull the required repos
-Alert user if repo is out of date/on non standard branch
-Deploy latest to display node

FuriousJulius, 2017-10-04:
finished here: https://github.com/EndPointCorp/lg_ros_nodes/commit/aac1113a7d9c92e99474152c6dc6daac4d7d996e

closed 2017-10-04 by FuriousJulius

minshallj, 2017-10-04:
possibly fixed. maybe @kiselev-dv could try it out? Just pull the latest from master

wzin, 2017-10-05:
[image]