EndPointCorp/lg_ros_nodes #381: Make Logitech Keyboard compatible with display nodes other than head-node (42-a)

Jbuhgit, 2018-05-30:
**Overview:**
The Logitech keyboard that is currently being tested on ISO version 2.11.0-rc15. The keyboard is functional however the cursor that appears when using the Touchpad on the keyboard, only allows the cursor to exist on 42-a nodes (tested @ ephq) . 

**Requirements:**
List of requirements that need to be satisfied that are needed to consider this feature implemented.
- Cursor should be able to move from L1 / R1 to other displays setup on other display nodes (42-b)
- Cursor is able to navigate across all 7 screens.
- Cursor does not get stuck on 3 center screens.

**Development notes**
Optional - list of places to look into and general guidelines (patterns to use, instructions and helpers)

**Testing scenario**
- Using touchpad on Logitech keyboard , test whether the cursor is able to move across all screens on  LG.

If this task is testable - include list of commands and expected outputs.

jandro-air, 2019-06-26:
Looks like this continues to be an issue. Following case:

User attempts to use Logitech keyboard to access a webpage on the outer 4 screens of an LGX (42-b). They cannot, because the cursor has been limited to the center three screens (42-a). The cursor stops at the borders of the R1 and L1. 

We have verified that systems that experience this are using the correct ros nodes. It would be good to see if we can prevent this from happening at all to provide a more seamless experience.

jandro-air, 2019-07-30:
allegedly resolved by @FuriousJulius and @mvollrath on bionic iso