EndPointCorp/lg_ros_nodes #383: user data dir constant

minshallj, 2018-06-01:
So we currently remove and create a new user data dir every time the chrome browser dies (so it's in the respawn as well as the startup of the chrome browser)

We want to be able to pass an extra option via director message that says "use this user data dir" and has a path to it. If it's not a directory already, create the directory, otherwise do nothing. Make sure we don't delete it on relaunches or scene changes. Watch out for soft relaunches too, they probably just call the respawn part, but watch out. A good way to manually test that this works is log into some web page, change scenes and go back and make sure you're still logged in.

minshallj, 2018-06-08:
https://github.com/EndPointCorp/lg_ros_nodes/blob/246d9dc0cb8d7a6c44e74f10c3a4fa6a41c16c8c/lg_common/src/lg_common/managed_browser.py#L159

init_tmp_dir should only be run the first time, so if the directory isn't created yet, then initialize it. Also only add the respawn handlers if we don't have a self.user_data_dir (or if it is not None) https://github.com/EndPointCorp/lg_ros_nodes/blob/master/lg_common/src/lg_common/managed_browser.py#L155

Managed browsers are mostly used as ManagedAdhocBrowsers, which are created via adhocbrowserpool 

In ManagedAdhocBrowser you just need to give it the user_data_dir init variable, and pass that along to our super class
https://github.com/EndPointCorp/lg_ros_nodes/blob/master/lg_common/src/lg_common/managed_adhoc_browser.py#L15

In the pool, you should check for the user_data_dir argument as part of the director message.
https://github.com/EndPointCorp/lg_ros_nodes/blob/e6e0941d510cd9b042c4a04583292f9b1d77da9a/lg_common/src/lg_common/adhoc_browser_pool.py#L339
Around there, you need to grab out the user_data_dir part of the message and pass it to the managedBorwser init method.

FuriousJulius, 2018-12-11:
This is in lg_ros_nodes [1.18.21](https://github.com/EndPointCorp/lg_ros_nodes/tree/1.18.21)

closed 2018-12-11 by FuriousJulius