EndPointCorp/lg_ros_nodes #399: Video sync feature has some stuttering

jandro-air, 2019-06-06:
**Overview:**
We use mplayer's udp sync feature to sync videos so that they play across the full width of the LG (i.e. across 42-a and b).

We somehow only ever tested this using videos that fade in from black :sweat_smile: . Turns out that there is a fair amount of stutter that happens at the beginning of the video during a loop while the slaves attempt to reconnect to the master. In other words, there is no stutter with the initial playback, but there is stutter once the video ends and repeats.

Would be great if there was some way we could smooth this out.

**Requirements:**
- No stutter on video loop

**Development notes**

One thought that comes to mind is somehow delaying playback upon video repeat until the slaves connect with the master. Another thought is that this stutter is related to how the video is told to repeat (i.e. the video CMS asset type using the `repeat` attribute) and not related to mplayer itself.

**Testing scenario**

Build a scene using a video that uses both the sync feature and is set to repeat. Let play on an LG. Notice how it loops but 42-b stutters when the video repeats. An example scene can be found here: https://ephq.galaxy.endpoint.com/#/group/97/presentation/841/scene/10482

JonathanPerlin, 2019-06-06:
Video of Stutter

jandro-air, 2019-06-11:
Some discussion with the R+D team has led to @mvollrath suggesting use of a new video player, found here: https://github.com/EndPointCorp/lg_ros_nodes/tree/master/lg_sv/webapps/videosync

Next steps would be to build some sort of launcher for this video player and test it out on an LG

mvollrath, 2019-07-09:
A GStreamer-based synchronized video player is now in lg_media.  It replaces the MPlayer launcher.